chipsalliance/chisel #3977: SRAM blackbox substitution options needed in Chisel 6.2.0

Gwins7, 2024-04-09:
**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**
I'm trying to use Nutshell (https://github.com/OSCPU/NutShell/) compiled by Chisel 6.2.0, and boot linux kernel using it on Xilinx Pynq-z2 board. The SRAM arrays generated by Chisel itself somehow do not function properly, and cause read-0-operation on board, which leads to boot fault. In the former version of Nutshell (compiled by Chisel 3), there are "--infer-rw" and "--repl-seq-mem" options in Makefile to turn SRAM into Blackboxs and provide .conf file. A python script ("./scripts/vlsi_mem_gen") given by RocketChip (https://github.com/chipsalliance/rocket-chip/) is used to generate specific SRAM verilog code according to .conf file. The Makefile attach the code at the end of generated verilog file, and the substitution completes. However, the "--infer-rw" and "--repl-seq-mem" option is deprecated in Chisel 6.
**Describe the solution you'd like**
I want to do the SRAM substitution automatically (by changing options or sth.), so extra options (new or old) are needed.
**Describe alternatives you've considered**
I have tried substituting Chisel-6-generated array with vlsi_mem_gen's array manually, and it works.
**What is the use case for implementing this feature?**
When user needs to substitute SRAM's verilog implementation according to actual board specification using external scripts, they'll need these options.

sequencer, 2024-04-09:
Here are some points:
- try to migrate to SRAM API?
- please use CIRCT to Replace the memory block.

And please note: RocketChip vlsi-mem-gen is deprecated and removed.

But actually, the main issue is the memory behavior model is a hard macro, while Chisel only generate a "standard" behavior model to it. The real flow should be using this model to wrap foundry RAM or FPGA RAM.

Some changes to chisel should also be considered:
- Deprecate `Mem` and `SyncMem` in Chisel, switching to `SRAM` and `ROM` API for hard macros.
- Since we have `ModuleChoice` API, we should consider use macro to generate multiple memory(ASIC Blackbox, FPGA BRAM, simulation) for a same Verilog

Gwins7, 2024-04-09:
@sequencer, Thank you for answering my question!
I've tried to use CIRCT option directly, but they seems to deprecate these options as well.
As for the migration to SRAM API, because of the inadequacy of Chisel 6 document (to me, who is a Chisel rookie) and the high integration level of the origin project, that is kind of hard. But if that is the only solution, I have no choice but to do so. I would be appreciated if you could give me more advice to finish this task.
Anyway, thank you! :)

jackkoenig, 2024-04-10:
firtool does support `--repl-seq-mem` and `--repl-seq-mem-file` so you should still be able to use those options. I don't see `--infer-rw` anymore but I think it's just always enabled.

I would like to understand more about:

> The SRAM arrays generated by Chisel itself somehow do not function properly, and cause read-0-operation on board, which leads to boot fault.

Do you know why this is the case? The SRAM arrays should be functionally correct so this is surprising to hear. If you can provide more information about what is functionally incorrect that would be very helpful.

Something to look out for is the infamous `Synth 8-6430` INFO[[1](https://www.beyond-circuits.com/wordpress/2019/10/ram-address-conflicts-and-a-vivado-synthesis-bug/)]where Vivado just changes the semantics of your memory (to make it use fewer resources) and all you get is s stupid info. firtool might be emitting Verilog that avoids this with `(* rw_addr_collision = "yes" *)`, but that's something to check for in your synthesis logs.

Gwins7, 2024-04-11:
@jackkoenig, Thank you for answering my question!

I searched the `-repl-seq-mem` information in CIRCT repository again carefully after your notification, then I found this:

[[Dedup][FIRRTL] SRAM modules not deduped when using memory macro replacement #6445](https://github.com/llvm/circt/issues/6445)

According to the code pattern in this issue, `-repl-seq-mem --repl-seq-mem-file=ext_mem.conf` should be used to generate .conf file. I originally thought that the `ext_mem.conf` is a file generated before Chisel compilation, but this time after tries I realized that this is just the name of compile-generated .conf file. After add these options in Firtool option, I can get .conf file content in .v file (because I don't use `--split-verilog` option), and all I have to do is to write script in Makefile to edit it. Problem solved.

About your last question, I think it is SRAM 's issue because I encountered a similar problem in another FPGA project. So, I just gave the substitution solution a try, and it solved the boot fault. What you're saying is actually possible. Because the Pynq board isn't available for me currently, I can not give you my test result here. I can send this to you later if you are interested.

closed 2024-04-11 by Gwins7