chipsalliance/chisel #4004: mill format check failing in CI

dtzSiFive, 2024-04-17:
https://github.com/chipsalliance/chisel/actions/runs/8720776315/job/23923064096

Also in the firtool bump, similarly: https://github.com/chipsalliance/chisel/pull/4002 .

However when I run the mill check or reformat commands locally things are clean.

Can someone take a look? Thanks!

dtzSiFive, 2024-04-17:
Oh! Maybe because that branch has commits not on main!

https://github.com/chipsalliance/chisel/commit/2d0f65ef52fc39a5e781ca891594d09d5c8131e8

dtzSiFive, 2024-04-17:
https://github.com/chipsalliance/chisel/pull/4002/commits/6cd00cdfc7bcfd8a5dc48af143d0879cff643c8d ?

jackkoenig, 2024-04-17:
We just switched from sbt format to mill format (like a few days ago) and the change included now formatting more stuff. Perhaps the particular incorrectly formatting change predated that switch to mill format so slipped through the cracks?

closed 2024-04-18 by dtzSiFive

seldridge, 2024-04-18:
Thanks for getting this sorted out. 💯