chipsalliance/chisel #4051: provide an Chisel Native API for SystemVerilog bind

sequencer, 2024-05-03:
**Type of issue**: Feature Request

For a `FixedIOModule`, the `ioGenerator` is its header.

Think about this scenario, Different teams can only access the header for different purpose, e.g. RTL designer, DV engineer seeing `Probe` and `ProbeRW`, Performance Model engineer writing necessary `printf` to specific signals(`Probe` it).

There will be three different end users to the specified `ioGenerator`, and three layers of IO in three colors. three teams are developing three Modules but bind at last.

Currently `bind` can be automatically extracted in CIRCT.

But we may need a way to expose this functionality in Chisel.

jackkoenig, 2024-05-03:
This is actually what Layers are: https://www.chisel-lang.org/docs/explanations/layers

sequencer, 2024-05-03:
Yes. Layer is doing good in firrtl spec. But at chisel level, I’m thinking is there some design pattern for it? e.g. aop for a same IO, and corresponding aspect will have corresponding layer. This is possible. But does it support D/I? I’m not sure.