chipsalliance/chisel #4128: PeekPokeAPI expectations don't return the actual test line (or I'm getting it very wrong)

kivikakk, 2024-06-02:
Hi all! I'm using ChiselSim with Chisel 6.4.0. The actual tests fine, but when an expectation fails, I sometimes find it hard to debug, except by adding loads of additional `message` arguments. For a concrete example:

```scala
it should "transmit a command packet" in {
  simulate(new SPI) { c =>
    c.reset.poke(true.B)
    c.clock.step()
    c.reset.poke(false.B)

    c.clock.step()
    c.io.req.ready.expect(true.B)

    c.io.req.bits.cmd.poke(0x19.U)
    c.io.req.bits.cmdRem.poke(0.U)
    c.io.req.bits.respLen.poke(0.U)
    c.io.req.valid.poke(true.B)

    c.clock.step()

    c.io.req.ready.expect(false.B)
  }
}
```

I haven't actually implemented any non-initial state yet, so I expect the final line to fail. And it does, but the spec file itself only gets a mention for the line with the `simulate` call (line 17 in this case). The full output is long, so it's hidden below.

<details><summary>`sbt test` output with `-oF` added to the test options</summary>

```sbt
sbt:ili9341spi> test
[info] SPISpec:
[info] SPI
[info] - should transmit a command packet *** FAILED ***
[info]   chisel3.simulator.PeekPokeAPI$FailedExpectationException: Failed Expectation: Observed value '1' != 0. Expectation failed: observed value 1 != 0
[info]   at chisel3.simulator.PeekPokeAPI$testableData.$anonfun$expect$9(PeekPokeAPI.scala:135)
[info]   at chisel3.simulator.PeekPokeAPI$testableData.$anonfun$expect$9$adapted(PeekPokeAPI.scala:133)
[info]   at svsim.Simulation$Port$$anonfun$check$1.applyOrElse(Simulation.scala:426)
[info]   at svsim.Simulation$Port$$anonfun$check$1.applyOrElse(Simulation.scala:424)
[info]   at scala.runtime.AbstractPartialFunction.apply(AbstractPartialFunction.scala:35)
[info]   at svsim.Simulation$Controller.$anonfun$completeInFlightCommands$1(Simulation.scala:195)
[info]   at svsim.Simulation$Controller.$anonfun$completeInFlightCommands$1$adapted(Simulation.scala:192)
[info]   at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:576)
[info]   at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:574)
[info]   at scala.collection.AbstractIterable.foreach(Iterable.scala:933)
[info]   at svsim.Simulation$Controller.completeInFlightCommands(Simulation.scala:192)
[info]   at chisel3.simulator.package$AnySimulatedModule.completeSimulation(package.scala:56)
[info]   at chisel3.simulator.Simulator.$anonfun$_simulate$1(Simulator.scala:112)
[info]   at chisel3.simulator.package$ChiselSimulation.$anonfun$runElaboratedModule$2(package.scala:97)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[info]   at chisel3.simulator.package$AnySimulatedModule$.withValue(package.scala:81)
[info]   at chisel3.simulator.package$ChiselSimulation.$anonfun$runElaboratedModule$1(package.scala:97)
[info]   at svsim.Simulation.$anonfun$run$3(Simulation.scala:51)
[info]   at scala.util.Try$.apply(Try.scala:210)
[info]   at svsim.Simulation.run(Simulation.scala:50)
[info]   at chisel3.simulator.package$ChiselSimulation.runElaboratedModule(package.scala:94)
[info]   at chisel3.simulator.Simulator$WorkspaceCompiler.$anonfun$process$1(Simulator.scala:64)
[info]   at scala.util.Try$.apply(Try.scala:210)
[info]   at chisel3.simulator.Simulator$WorkspaceCompiler.liftedTree1$1(Simulator.scala:64)
[info]   at chisel3.simulator.Simulator$WorkspaceCompiler.process(Simulator.scala:53)
[info]   at chisel3.simulator.SingleBackendSimulator.processBackends(Simulator.scala:139)
[info]   at chisel3.simulator.SingleBackendSimulator.processBackends$(Simulator.scala:138)
[info]   at chisel3.simulator.EphemeralSimulator$DefaultSimulator.processBackends(EphemeralSimulator.scala:28)
[info]   at chisel3.simulator.Simulator._simulate(Simulator.scala:116)
[info]   at chisel3.simulator.Simulator._simulate$(Simulator.scala:97)
[info]   at chisel3.simulator.EphemeralSimulator$DefaultSimulator._simulate(EphemeralSimulator.scala:28)
[info]   at chisel3.simulator.SingleBackendSimulator.simulate(Simulator.scala:146)
[info]   at chisel3.simulator.SingleBackendSimulator.simulate$(Simulator.scala:142)
[info]   at chisel3.simulator.EphemeralSimulator$DefaultSimulator.simulate(EphemeralSimulator.scala:28)
[info]   at chisel3.simulator.EphemeralSimulator$.simulate(EphemeralSimulator.scala:25)
[info]   at ee.kivikakk.ili9341spi.spi.SPISpec.$anonfun$new$1(SPISpec.scala:17)
[... 62 lines cut ...]
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed tests:
[error]         ee.kivikakk.ili9341spi.spi.SPISpec
[error] (Test / test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 3 s, completed Jun 2, 2024, 4:12:17 PM
sbt:ili9341spi>
```

</details>

```sbt
sbt:ili9341spi> lastGrep SPISpec test
[debug] Running TaskDef(ee.kivikakk.ili9341spi.spi.SPISpec, org.scalatest.tools.Framework$$anon$1@309d642a, false, [SuiteSelector])
[info] SPISpec:
[info]   at ee.kivikakk.ili9341spi.spi.SPISpec.$anonfun$new$1(SPISpec.scala:17)
[error]         ee.kivikakk.ili9341spi.spi.SPISpec
sbt:ili9341spi>
```

I'm sure this is hard to avoid because `expect` is probably just enqueuing the expectation until the sim runtime next goes back to talk to the simulator. (Or something — I haven't looked into how it works at all, other than seeing that such a flow is possible!)

I guess it'd be nice to also capture `SourceInfo` when doing an `expect`, and to report it with any failures? Any thoughts, or if I'm missing something big?

kivikakk, 2024-06-02:
I should add — I'm happy to have a go at the implementation if it is indeed as simple a matter as I imagine!

jackkoenig, 2024-06-03:
Thanks for the report! Improving this would be really good for the user experience.

There are two ways we could fix it:
1. Grab a SourceInfo and leverage Chisel's error reporting mechanisms to report the error (can even include source line and carat for most cases)
2. Grab a stack trace and throw an exception pointing to the line

(1) probably the easiest and definitely the most performant, (2) grabbing stack traces for every single expect could be a bit expensive (more expensive than simulation? unclear, would need to be measured).

Thank you for your interest and having a go at the implementation. If you can get to it, great, if not, at least this will serve as a suggested implementation for anyone else who gets a chance to take a look.

You can capture a `SourceInfo` by adding it as an implicit argument to the method, in this case `expect`[^1] (note that every overloaded method will need it). I think you just need then to use this SourceInfo in reporting the error where there is currently an exception thrown[^2].

If you just want to report the source locator, that should be enough. If you want to make it a bit prettier with Chisel pulling the source-line and having a carat pointing to the offending expect, you can try using some of Chisel's error reporting logic in the ErrorLog[^3]. You may need to massage the internal APIs a bit as they are very much built around the assumption of a Chisel runtime that reports errors at the end (which SVSim does not interact with at all). Not sure it's worth it for you to attempt that, but it's an option!

[^1]: https://github.com/chipsalliance/chisel/blob/0f2efb52a78be5484fa231d5d635ce0a4b124bd8/src/main/scala/chisel3/simulator/PeekPokeAPI.scala#L59
[^2]: https://github.com/chipsalliance/chisel/blob/0f2efb52a78be5484fa231d5d635ce0a4b124bd8/src/main/scala/chisel3/simulator/PeekPokeAPI.scala#L135
[^3]: https://github.com/chipsalliance/chisel/blob/0f2efb52a78be5484fa231d5d635ce0a4b124bd8/core/src/main/scala/chisel3/internal/Error.scala#L251

kivikakk, 2024-06-04:
Thank you so much for the pointers! I'll get (1) going as-is to have something minimal, and then I'll give the better reporting a try. If I have the time I'll see about (2); I can imagine it might indeed be expensive.

closed 2024-06-06 by jackkoenig

jackkoenig, 2024-06-06:
This has been merged to `main` and backported to `6.x`. I'm not sure when `6.5.0` will be released with this improvement but you can use a SNAPSHOT that includes it `6.4.0+3-26ae5a85-SNAPSHOT`. You'll need to enable the resolver for SonatypeS01 Snapshots to use that, instructions vary by build tool:
* Scala-Cli: `//> using repository "sonatype-s01:snapshots"`
* SBT: `resolvers += Resolver.sonatypeOssRepos("snapshots")`

kivikakk, 2024-06-07:
Ta! 😊

PR #4144 (2024-06-06): PeekPokeAPI: include source location on failed expect() calls.
Fixes #4128. (cc @jackkoenig)

Some notes:

* b3fca5e adds the `SourceInfo` collection and reporting in the exception.
  * I haven't tried stack trace collection. Other than simply not getting to it yet, I don't think I have a rigorous enough setup (or a large enough testcase, which I have a working build going for) to meaningfully assess the performance cost.
  * Minor API design question: do we ever expect `testableData.expect` (with `expected`, `encode` and `buildMessage` arguments) to be called by a test user directly? If not,  [... cut]
PR #4149 (2024-06-06): PeekPokeAPI: include source location on failed expect() calls. (backport #4144)
Fixes #4128. (cc @jackkoenig)

Some notes:

* b3fca5e adds the `SourceInfo` collection and reporting in the exception.
  * I haven't tried stack trace collection. Other than simply not getting to it yet, I don't think I have a rigorous enough setup (or a large enough testcase, which I have a working build going for) to meaningfully assess the performance cost.
  * Minor API design question: do we ever expect `testableData.expect` (with `expected`, `encode` and `buildMessage` arguments) to be called by a test user d [... cut]