chipsalliance/chisel #4208: [ChiselSim] ChiselEnum does not provide peek/poke methods and user-friendly API

carlosedp, 2024-06-23:
While rewriting some tests from chiseltest to ChiselSim, I've noticed ChiselEnum is not handled by the Peek/Poke API.

In a test with `c.io.inst.peek() should be(inst)` where `inst` is a `ChiselEnum.Type`, I get: `value peek is not a member of chiselv.Instruction.Type`.

The closest thing I could get to work is

```scala
c.io.inst.peekValue().asBigInt should be(inst.litValue)
```

Which is very convoluted and not user-friendly.

kammoh, 2024-06-24:
Would [this](https://github.com/chipsalliance/chisel/blob/9f148207b0f783a61fa61ffe0dad4fb66211e9e1/src/test/scala/chiselTests/simulator/PeekPokeAPISpec.scala#L48) be the API you're looking for?

carlosedp, 2024-06-24:
Having peek() on ChiselEnum would be nice... but it's not in merged into Chisel, right?

jackkoenig, 2024-06-24:
> Having peek() on ChiselEnum would be nice... but it's not in merged into Chisel, right?

Not yet, but @kammoh has opened a draft PR adding it (among many other improvements): https://github.com/chipsalliance/chisel/pull/4209

I am reviewing, but got sick over the weekend so am not quite at 100% 🙃

carlosedp, 2024-06-24:
Oh, nice! just saw it mentioned! Commented here before I saw it.

Take care my friend! Hope you get better soon :)