chipsalliance/chisel #4240: Please make some classes in simulator package public for simulator customization.

zhutmost, 2024-07-02:
**Type of issue**: Feature Request

**Is your feature request related to a problem? Please describe.**
I want to customize the simulation process with svsim. But some important classes in [chisel3/simulator/package] are `private`, such as:
- `class SimulatedModule` has a private constructor. It should be public, or I cannot create one during simulation.
- `object AnySimulatedModule` is private. please make it public. Or users cannot use its methods `withValue` and `current`.
- `AnySimulatedModule` is sealed (cannot be extended) and has many private methods.

I am writing some new things like [`PeekPokeAPI`](https://github.com/chipsalliance/chisel/blob/main/src/main/scala/chisel3/simulator/PeekPokeAPI.scala) and [`EphemeralSimulator`](https://github.com/chipsalliance/chisel/blob/main/src/main/scala/chisel3/simulator/EphemeralSimulator.scala). And the above APIs are private so that I cannot access them (unless I fork a whole chisel).

**Describe the solution you'd like**
make the above simulator's APIs public.

**Describe alternatives you've considered**
N/A.

**Additional context**
N/A.

**What is the use case for implementing this feature?**

Emin017, 2025-07-05:
After the release of Chisel7-RC2, we have more methods for setting up simulators. Please check [ChiselSimSpec.scala](https://github.com/chipsalliance/chisel/blob/v7.0.0-RC2/src/test/scala-2/chiselTests/simulator/scalatest/ChiselSimSpec.scala) for more usage examples.

You can also search for the keyword "**ChiselSim**" in [pull requests](https://github.com/chipsalliance/chisel/pulls?q=ChiselSim) to find recent work on ChiselSim/Svsim.

zhutmost, 2025-07-06:
@Emin017 Thanks a lot!
If we could have a detailed document about the new ChiselSim, it would be better.

Let me close this issue first. Again thanks for developers' great jobs!

Emin017, 2025-07-06:
> [@Emin017](https://github.com/Emin017) Thanks a lot! If we could have a detailed document about the new ChiselSim, it would be better.
> 
> Let me close this issue first. Again thanks for developers' great jobs!

see: https://www.chisel-lang.org/docs/explanations/testing#chiselsim

Emin017, 2025-07-09:
cc @seldridge