chipsalliance/chisel #4246: ChiselSim waveforms

HakamAtassi, 2024-07-07:
Hello!

I feel like I'm not the first to ask, but I haven't found anything regarding this as an open issue so I'll ask nonetheless. 

ChiselSim. Waveforms. Yes or no? 

Can I write chisel/scala tests and dump a VCD like ChiselTest?

zhutmost, 2024-07-08:
A simple answer is: yes

HakamAtassi, 2024-07-08:
> A simple answer is: yes

How? Is there a native way of doing that or does it involve tywaves?

zhutmost, 2024-07-08:
Enable `traceEnabled` when you`runElaboratedModule` during simulation. See https://github.com/chipsalliance/chisel/blob/8f00067a973ce377440bf16b9a49a4120316217b/src/main/scala/chisel3/simulator/package.scala#L92

And I have [some wrapper code](https://github.com/zhutmost/chipmunk/blob/fc1877b713a3f59713bc3f652b447fa68913c403/chipmunk/src/tester/TestRunner.scala#L35) to make it much easier. See my example [here](https://github.com/zhutmost/chipmunk/blob/main/chipmunk/test/src/TestRunnerSpec.scala).

jackkoenig, 2024-07-09:
It's not exposed with the existing EphemeralSimulator, but there is work towards simpler user APIs like https://github.com/chipsalliance/chisel/pull/4209 which should make this easier.

edwardcwang, 2024-11-01:
On vanilla Chisel 6.5 this ended up requiring two separate hacks against `EphemeralSimulator`:

1) https://github.com/edwardcwang/decoupled-serializer/blob/master/src/main/scala/VCDHackedEphemeralSimulator.scala#L44
Inserting `traceStyle = Some(verilator.Backend.CompilationSettings.TraceStyle.Vcd(traceUnderscore = true))` into the `backendSpecificCompilationSettings` of the `DefaultSimulator`.

2) https://github.com/edwardcwang/decoupled-serializer/blob/master/src/main/scala/VCDHackedEphemeralSimulator.scala#L31
The above was not enough as it added support for tracing but did not actually call the functions to enable & start the dumping.
Hacking the `simulate()` method to insert a call to `module.controller.setTraceEnabled(true)` to actually enable the dump of the `.vcd` file.

Third bonus hack: use the local `test_run_dir` directory (from older versions of Chisel) instead of a temporary one which gets deleted.
https://github.com/edwardcwang/decoupled-serializer/blob/master/src/main/scala/VCDHackedEphemeralSimulator.scala#L61