chipsalliance/chisel #4293: Scala version

schoeberl, 2024-07-19:
**Type of issue**: Documentation missing

**Please provide the steps to reproduce the problem:**

Try to use Chisel 3.5.6 with Scala 2.13.14 (and many other combinations)

**What is the current behavior?**

missing chisel3-plugin_2.13.14

As a result, Chisel depends on a fully specified Scala version.

We should document which Scala version is supported for which Chisel (plugin) version.

schoeberl, 2024-07-19:
Maybe it is enough to just show the maximum version, or a link to the Maven pages, e.g., https://mvnrepository.com/artifact/edu.berkeley.cs/chisel3-plugin

schoeberl, 2024-11-01:
I have tried to write up that info here: https://github.com/schoeberl/chisel-book/blob/63986db5aafdfaa07d58e7a1ff680cf9a44db298/chisel-book.tex#L1711-L1741

Is this OK?

Yakkhini, 2025-12-16:
> I have tried to write up that info here: https://github.com/schoeberl/chisel-book/blob/63986db5aafdfaa07d58e7a1ff680cf9a44db298/chisel-book.tex#L1711-L1741
> 
> Is this OK?

That's great! Thanks your work on this book, which help me a lot when learning Chisel.