chipsalliance/chisel #4305: UInt<1>(0)' must be a Chisel type, not hardware

mlabaf2, 2024-07-24:
Hi
I defined class  cap_result_t , I use this class in another class  crevokeModule . then use crevokeModule  in another class by this sentence "  val crevokeModule = Module(new crevokeModule(new cap_result_t))". 
I do not know why I get below error after using  crevokeModule  ?!

error
"Exception in thread "main" chisel3.package$ExpectedChiselTypeException: 'UInt<1>(0)' must be a Chisel type, not hardware" 

class  cap_result_t extends Bundle{
    val cap   = clen_t
    val tag   = UInt (1.W)
    val valid = UInt (1.W)
   }

 class crevokeModule ( typ: cap_result_t) extends Module {

//  class crevokeModule extends bundle {
   val io = IO(new Bundle() {
   val src = Input(UInt(XLEN.W))
   val crevoke = Output(UInt(XLEN.W))
  })

val rd_result_o = IO(Output(typ))

 rd_result_o := 0.U.asTypeOf(chiselTypeOf(rd_result_o))
 rd_result_o.tag:= 1.U(1.W)

}

jackkoenig, 2024-07-25:
There's not enough in the message to see what the error is, but I'm guessing there is something wrong with `clen_t`, where is it defined?

It would be easiest to help you if you can provide a reproducible example. Ideally this would be in the form of a stand-alone Scala CLI example, see https://www.chisel-lang.org/docs/installation#quickstart-with-scala-cli for instructions on how to set this up, then you can copy-paste the full, executable code example into the issue. Alternatively you can use Scastie which works in your browser, you can write your code in the template: https://scastie.scala-lang.org/gJse8Wv6RsGX4inwTjjGQA, and then share the link after running it and showing the error you are seeing. The only issue with Scastie is it can't generate Verilog, but it can be used to show basic issues.

mlabaf2, 2024-07-26:
yes the problem was clen_t.
thanks

closed 2024-09-06 by jackkoenig