chipsalliance/chisel #4328: How to get the current simulated clock time?

reoLantern, 2024-08-04:
I hope to use printf to print the simulation cycle or specific nanoseconds, but there seems to be no corresponding function in the chisel api. I don't mean `timestamp`, which would print the system time.

sequencer, 2024-08-04:
This is out of Chisel's capability, Here are something you may need refer to: 
- [SystemVerilog LRM](https://standards.ieee.org/ieee/1800/7743/) 
- [vpi_user.h](https://github.com/verilator/verilator/blob/master/include/vltstd/vpi_user.h)
- [svdpi.h](https://github.com/verilator/verilator/blob/master/include/vltstd/svdpi.h)

You can acquire anything you wish with chisel DPI, and use `printf` to print the result out. Or wrap it with blackbox encoding a `$system` call.

closed 2024-08-04 by sequencer

uenoku, 2024-08-05:
There has been some work on emitting the current simulation time from Chisel `printf`
CC: @girishpai @seldridge