chipsalliance/chisel #4354: 2d Vec RegInit initailized randomly

thuako, 2024-08-16:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

**What is the current behavior?**

If I generate RTL with '--preserve-aggregate=all', a register that is of the RegInit 2dimension Vector object type in Chisel is randomly initialized in RTL.

chisel
```scala
import chisel3._
class MybundleVec extends Bundle{
  val in_vec = Input(Vec(16, UInt(16.W)))
  val out_vec = Output(Vec(16, UInt(16.W)))
}

class AggregateMode extends Module {
  val io = IO(new MybundleVec())

  val my_queue = RegInit(0.U.asTypeOf(Vec(4, chiselTypeOf(io.out_vec))))

  my_queue.foldLeft(io.in_vec){(source, dest) => dest <> source; dest}
  io.out_vec := my_queue.last
}

object Generator extends App {
  ChiselStage.emitSystemVerilogFile(
    new AggregateMode(),
    args = Array("-ll", "debug", "--target-dir", "test_emit"),
    firtoolOpts = Array(
    "-strip-debug-info",
    "--disable-annotation-unknown",
    s"--include-dir=${Paths.get("src/main/resources").toAbsolutePath.toString}",
    "--emit-chisel-asserts-as-sva",
    "--emit-omir",
    "--strip-fir-debug-info",
    "--lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,disallowPortDeclSharing,locationInfoStyle=wrapInAtSquareBracket",
    "--mlir-timing",
    "-O=release",
    "--preserve-aggregate=all",
    "--scalarize-public-modules=false",
    "--scalarize-ext-modules=false",
    ),
  )
}
```

generated RTL
```verilog
module AggregateMode(
  input               clock,
  input               reset,
  input  [15:0][15:0] io_in_vec,
  output [15:0][15:0] io_out_vec
);

  reg [3:0][15:0][15:0] my_queue;
  always @(posedge clock) begin
    my_queue[2'h0] <= io_in_vec;
    my_queue[2'h1 +: 3] <= my_queue[2'h0 +: 3];
  end // always @(posedge)
  `ifdef ENABLE_INITIAL_REG_
    `ifdef FIRRTL_BEFORE_INITIAL
      `FIRRTL_BEFORE_INITIAL
    `endif // FIRRTL_BEFORE_INITIAL
    logic [31:0] _RANDOM[0:31];
    initial begin
      `ifdef INIT_RANDOM_PROLOG_
        `INIT_RANDOM_PROLOG_
      `endif // INIT_RANDOM_PROLOG_
      `ifdef RANDOMIZE_REG_INIT
        for (logic [5:0] i = 6'h0; i < 6'h20; i += 6'h1) begin
          _RANDOM[i[4:0]] = `RANDOM;
        end
        my_queue[2'h0][4'h0] = _RANDOM[5'h0][31:16];
        my_queue[2'h0][4'h1] = _RANDOM[5'h0][15:0];
        my_queue[2'h0][4'h2] = _RANDOM[5'h1][31:16];
        my_queue[2'h0][4'h3] = _RANDOM[5'h1][15:0];
        my_queue[2'h0][4'h4] = _RANDOM[5'h2][31:16];
        my_queue[2'h0][4'h5] = _RANDOM[5'h2][15:0];
        my_queue[2'h0][4'h6] = _RANDOM[5'h3][31:16];
        my_queue[2'h0][4'h7] = _RANDOM[5'h3][15:0];
        my_queue[2'h0][4'h8] = _RANDOM[5'h4][31:16];
        my_queue[2'h0][4'h9] = _RANDOM[5'h4][15:0];
        my_queue[2'h0][4'hA] = _RANDOM[5'h5][31:16];
        my_queue[2'h0][4'hB] = _RANDOM[5'h5][15:0];
        my_queue[2'h0][4'hC] = _RANDOM[5'h6][31:16];
        my_queue[2'h0][4'hD] = _RANDOM[5'h6][15:0];
        my_queue[2'h0][4'hE] = _RANDOM[5'h7][31:16];
        my_queue[2'h0][4'hF] = _RANDOM[5'h7][15:0];
        my_queue[2'h1][4'h0] = _RANDOM[5'h8][31:16];
        my_queue[2'h1][4'h1] = _RANDOM[5'h8][15:0];
        my_queue[2'h1][4'h2] = _RANDOM[5'h9][31:16];
        my_queue[2'h1][4'h3] = _RANDOM[5'h9][15:0];
        my_queue[2'h1][4'h4] = _RANDOM[5'hA][31:16];
        my_queue[2'h1][4'h5] = _RANDOM[5'hA][15:0];
        my_queue[2'h1][4'h6] = _RANDOM[5'hB][31:16];
        my_queue[2'h1][4'h7] = _RANDOM[5'hB][15:0];
        my_queue[2'h1][4'h8] = _RANDOM[5'hC][31:16];
        my_queue[2'h1][4'h9] = _RANDOM[5'hC][15:0];
        my_queue[2'h1][4'hA] = _RANDOM[5'hD][31:16];
        my_queue[2'h1][4'hB] = _RANDOM[5'hD][15:0];
        my_queue[2'h1][4'hC] = _RANDOM[5'hE][31:16];
        my_queue[2'h1][4'hD] = _RANDOM[5'hE][15:0];
        my_queue[2'h1][4'hE] = _RANDOM[5'hF][31:16];
        my_queue[2'h1][4'hF] = _RANDOM[5'hF][15:0];
        my_queue[2'h2][4'h0] = _RANDOM[5'h10][31:16];
        my_queue[2'h2][4'h1] = _RANDOM[5'h10][15:0];
        my_queue[2'h2][4'h2] = _RANDOM[5'h11][31:16];
        my_queue[2'h2][4'h3] = _RANDOM[5'h11][15:0];
        my_queue[2'h2][4'h4] = _RANDOM[5'h12][31:16];
        my_queue[2'h2][4'h5] = _RANDOM[5'h12][15:0];
        my_queue[2'h2][4'h6] = _RANDOM[5'h13][31:16];
        my_queue[2'h2][4'h7] = _RANDOM[5'h13][15:0];
        my_queue[2'h2][4'h8] = _RANDOM[5'h14][31:16];
        my_queue[2'h2][4'h9] = _RANDOM[5'h14][15:0];
        my_queue[2'h2][4'hA] = _RANDOM[5'h15][31:16];
        my_queue[2'h2][4'hB] = _RANDOM[5'h15][15:0];
        my_queue[2'h2][4'hC] = _RANDOM[5'h16][31:16];
        my_queue[2'h2][4'hD] = _RANDOM[5'h16][15:0];
        my_queue[2'h2][4'hE] = _RANDOM[5'h17][31:16];
        my_queue[2'h2][4'hF] = _RANDOM[5'h17][15:0];
        my_queue[2'h3][4'h0] = _RANDOM[5'h18][31:16];
        my_queue[2'h3][4'h1] = _RANDOM[5'h18][15:0];
        my_queue[2'h3][4'h2] = _RANDOM[5'h19][31:16];
        my_queue[2'h3][4'h3] = _RANDOM[5'h19][15:0];
        my_queue[2'h3][4'h4] = _RANDOM[5'h1A][31:16];
        my_queue[2'h3][4'h5] = _RANDOM[5'h1A][15:0];
        my_queue[2'h3][4'h6] = _RANDOM[5'h1B][31:16];
        my_queue[2'h3][4'h7] = _RANDOM[5'h1B][15:0];
        my_queue[2'h3][4'h8] = _RANDOM[5'h1C][31:16];
        my_queue[2'h3][4'h9] = _RANDOM[5'h1C][15:0];
        my_queue[2'h3][4'hA] = _RANDOM[5'h1D][31:16];
        my_queue[2'h3][4'hB] = _RANDOM[5'h1D][15:0];
        my_queue[2'h3][4'hC] = _RANDOM[5'h1E][31:16];
        my_queue[2'h3][4'hD] = _RANDOM[5'h1E][15:0];
        my_queue[2'h3][4'hE] = _RANDOM[5'h1F][31:16];
        my_queue[2'h3][4'hF] = _RANDOM[5'h1F][15:0];
      `endif // RANDOMIZE_REG_INIT
    end // initial
    `ifdef FIRRTL_AFTER_INITIAL
      `FIRRTL_AFTER_INITIAL
    `endif // FIRRTL_AFTER_INITIAL
  `endif // ENABLE_INITIAL_REG_
  assign io_out_vec = my_queue[2'h3];
endmodule

```

**What is the expected behavior?**

**Please tell us about your environment:**

**Other Information**

**What is the use case for changing the behavior?**

thuako, 2024-08-16:
And "--preserve-aggregate=none" compile option has also same bug.

kammoh, 2024-08-19:
What version of chisel are you using?
This is the output I see with Chisel `6.5.0` release, which seems to be correct.

```verilog

module AggregateMode(
  input         clock,
                reset,
  input  [15:0] io_in_vec_0,
                io_in_vec_1,
                io_in_vec_2,
                io_in_vec_3,
                io_in_vec_4,
                io_in_vec_5,
                io_in_vec_6,
                io_in_vec_7,
                io_in_vec_8,
                io_in_vec_9,
                io_in_vec_10,
                io_in_vec_11,
                io_in_vec_12,
                io_in_vec_13,
                io_in_vec_14,
                io_in_vec_15,
  output [15:0] io_out_vec_0,
                io_out_vec_1,
                io_out_vec_2,
                io_out_vec_3,
                io_out_vec_4,
                io_out_vec_5,
                io_out_vec_6,
                io_out_vec_7,
                io_out_vec_8,
                io_out_vec_9,
                io_out_vec_10,
                io_out_vec_11,
                io_out_vec_12,
                io_out_vec_13,
                io_out_vec_14,
                io_out_vec_15
);

  reg [15:0] my_queue_0_0;
  reg [15:0] my_queue_0_1;
  reg [15:0] my_queue_0_2;
  reg [15:0] my_queue_0_3;
  reg [15:0] my_queue_0_4;
  reg [15:0] my_queue_0_5;
  reg [15:0] my_queue_0_6;
  reg [15:0] my_queue_0_7;
  reg [15:0] my_queue_0_8;
  reg [15:0] my_queue_0_9;
  reg [15:0] my_queue_0_10;
  reg [15:0] my_queue_0_11;
  reg [15:0] my_queue_0_12;
  reg [15:0] my_queue_0_13;
  reg [15:0] my_queue_0_14;
  reg [15:0] my_queue_0_15;
  reg [15:0] my_queue_1_0;
  reg [15:0] my_queue_1_1;
  reg [15:0] my_queue_1_2;
  reg [15:0] my_queue_1_3;
  reg [15:0] my_queue_1_4;
  reg [15:0] my_queue_1_5;
  reg [15:0] my_queue_1_6;
  reg [15:0] my_queue_1_7;
  reg [15:0] my_queue_1_8;
  reg [15:0] my_queue_1_9;
  reg [15:0] my_queue_1_10;
  reg [15:0] my_queue_1_11;
[... 164 lines cut ...]
      my_queue_3_12 <= my_queue_2_12;
      my_queue_3_13 <= my_queue_2_13;
      my_queue_3_14 <= my_queue_2_14;
      my_queue_3_15 <= my_queue_2_15;
    end
  end // always @(posedge)
  assign io_out_vec_0 = my_queue_3_0;
  assign io_out_vec_1 = my_queue_3_1;
  assign io_out_vec_2 = my_queue_3_2;
  assign io_out_vec_3 = my_queue_3_3;
  assign io_out_vec_4 = my_queue_3_4;
  assign io_out_vec_5 = my_queue_3_5;
  assign io_out_vec_6 = my_queue_3_6;
  assign io_out_vec_7 = my_queue_3_7;
  assign io_out_vec_8 = my_queue_3_8;
  assign io_out_vec_9 = my_queue_3_9;
  assign io_out_vec_10 = my_queue_3_10;
  assign io_out_vec_11 = my_queue_3_11;
  assign io_out_vec_12 = my_queue_3_12;
  assign io_out_vec_13 = my_queue_3_13;
  assign io_out_vec_14 = my_queue_3_14;
  assign io_out_vec_15 = my_queue_3_15;
endmodule

```

thuako, 2024-08-20:
My chisel version is 6.5.0
And I think you didn't give '--preserve-aggregate=all' option.

In my code, my_queue defined as 2d vector register,
but in your code, my_queue defined as just register.

Could you generate with  '--preserve-aggregate=all' option ?

uenoku, 2024-08-20:
Thank you for the report. This looks like a bug of SFCCompat pass in CIRCT.

kammoh, 2024-08-20:
> My chisel version is 6.5.0 And I think you didn't give '--preserve-aggregate=all' option.
> 
> In my code, my_queue defined as 2d vector register, but in your code, my_queue defined as just register.
> 
> Could you generate with '--preserve-aggregate=all' option ?

Opps! Sorry I had missed that flag. Glad that @uenoku diagnosed the issue and has submitted a fix.

jackkoenig, 2024-08-20:
Thanks @uenoku! And thank you @thuako for the bug report!

closed 2024-09-03 by uenoku