chipsalliance/chisel #4418: Append layers to existing layer

unlsycn, 2024-09-27:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

```scala
object Verification extends Layer(LayerConfig.Extract()) {
  object BMC extends Layer(LayerConfig.Extract())
  object Debug extends Layer(LayerConfig.Extract())
}

addLayer(Verification.BMC)
addLayer(Verification.Debug)
```

**What is the current behavior?**

```firrtl
layer Verification, bind, "Verification" :
  layer BMC, bind, "Verification/BMC" :
  layer Debug, bind, "Verification/Debug" :
layer Verification, bind, "Verification" :
  layer Assert, bind, "Verification/Assert" :
  layer Assume, bind, "Verification/Assume" :
  layer Cover, bind, "Verification/Cover" :
```

which causes a firtool error `redefinition of symbol named 'Verification'`.

**What is the expected behavior?**

Children of a layer will be merged with an existing same layer.

seldridge, 2024-09-27:
I'm assuming that you want to nest the `BMC` and `Debug` layers under the upstream `Verification` layer, correct?

You can do this with the following:

``` scala
// The name of the object doesn't matter here.
object Verification {
  implicit val parent: Layer = chisel3.layers.Verification
  object BMC extends Layer(LayerConfig.Extract())
  object Debug extends Layer(LayerConfig.Extract())
}
```

Each `Layer` takes an implicit argument that is what it's parent layer is. You can override this by bringing an `implicit val` of type `Layer` into scope. (You can also pass it manually, but then you would also have to manually pass the source locators which does not work well.)

With the above, you then get:

```
FIRRTL version 4.0.0
circuit Foo :
  layer Verification, bind, "Verification" :
    layer BMC, bind, "Verification/BMC" :
    layer Debug, bind, "Verification/Debug" :
    layer Assert, bind, "Verification/Assert" :
    layer Assume, bind, "Verification/Assume" :
    layer Cover, bind, "Verification/Cover" :
```

unlsycn, 2024-09-27:
> I'm assuming that you want to nest the `BMC` and `Debug` layers under the upstream `Verification` layer, correct?
>
> You can do this with the following:
>
> ``` scala
> // The name of the object doesn't [...]

Thanks a lot! I'll give it a try.

closed 2024-09-28 by unlsycn