chipsalliance/chisel #4536: chisel3.util.Queue not working with SInt

jiegec, 2024-11-27:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

Run the following code:

```scala
import circt.stage.ChiselStage
import chisel3._
import chisel3.util._

class Test() extends Module {
  val io = IO(new Bundle {
    val a = Flipped(Decoupled(SInt(32.W)))
    val b = Decoupled(SInt(32.W))
  })
  val que = Module(new Queue(SInt(32.W), 1))
  que.io.enq.valid := io.a.valid
  que.io.enq.bits := io.a.bits
  io.a.ready := que.io.enq.ready

  io.b.bits := que.io.deq.bits + 1.S
  io.b.valid := que.io.deq.valid
  que.io.deq.ready := io.b.ready
}

object Test extends App {
  // These lines generate the Verilog output
  println(
    ChiselStage.emitSystemVerilog(
      new Test(),
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

**What is the current behavior?**

It fails with:

```
Exception in thread "main" circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: /home/jiegec/.cache/llvm-firtool/1.62.1/bin/firtool returned a non-zero exit code. Note that this version of Chisel (6.6.0) was published against firtool version 1.62.1.
------------------------------------------------------------------------------
ExitCode:
1
STDOUT:

STDERR:
src/main/scala/chisel3/util/Decoupled.scala:256:91: error: 'firrtl.strictconnect' op failed to verify that operands must be structurally equivalent
src/main/scala/chisel3/util/Decoupled.scala:256:91: note: see current operation: "firrtl.strictconnect"(%1, %15) : (!firrtl.sint<32>, !firrtl.uint<32>) -> ()

------------------------------------------------------------------------------
```

**What is the expected behavior?**

It should not fail

**Please tell us about your environment:**
    - version: `6.6.0`
    - OS: `Debian Bookworm`

unlsycn, 2024-11-27:
This is a CIRCT bug, the single-address Chisel memory is being substituted by reg and a incorrect type conversion in the process causes this problem. I will fix it tomorrow.

jackkoenig, 2024-11-27:
I can confirm this is still present in latest firtool (`1.95.1`):
```scala
//> using repository "sonatype-s01:snapshots"
//> using scala "2.13.15"
//> using dep "org.chipsalliance::chisel:7.0.0-M2+208-892fa802-SNAPSHOT"
//> using plugin "org.chipsalliance:::chisel-plugin:7.0.0-M2+208-892fa802-SNAPSHOT"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
import chisel3.util._
// _root_ disambiguates from package chisel3.util.circt if user imports chisel3.util._
import _root_.circt.stage.ChiselStage

class Test() extends Module {
  val io = IO(new Bundle {
    val a = Flipped(Decoupled(SInt(32.W)))
    val b = Decoupled(SInt(32.W))
  })
  val que = Module(new Queue(SInt(32.W), 1))
  que.io.enq.valid := io.a.valid
  que.io.enq.bits := io.a.bits
  io.a.ready := que.io.enq.ready

  io.b.bits := que.io.deq.bits + 1.S
  io.b.valid := que.io.deq.valid
  que.io.deq.ready := io.b.ready
}

object Test extends App {
  // These lines generate the Verilog output
  println(
    ChiselStage.emitSystemVerilog(
      new Test(),
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```
Gives:
```
Exception in thread "main" circt.stage.phases.Exceptions$FirtoolNonZeroExitCode: /Users/koenig/Library/Caches/org.chipsalliance.llvm-firtool/1.95.1/bin/firtool returned a non-zero exit code. Note that this version of Chisel (7.0.0-M2+208-892fa802-SNAPSHOT) was published against firtool version 1.95.1.
------------------------------------------------------------------------------
ExitCode:
1
STDOUT:

STDERR:
src/main/scala/chisel3/util/Decoupled.scala:256:91: error: 'firrtl.matchingconnect' op failed to verify that operands must be structurally equivalent
src/main/scala/chisel3/util/Decoupled.scala:256:91: note: see current operation: "firrtl.matchingconnect"(%1, %15) : (!firrtl.sint<32>, !firrtl.uint<32>) -> ()

------------------------------------------------------------------------------
```

seldridge, 2024-11-27:
CC: @rwy7, @youngar as you all were just recently looking at that canonicalizer.

closed 2025-08-21 by jiegec