chipsalliance/chisel #4541: Hierarchy Cookbook example not working

jiegec, 2024-11-28:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

Save the following code in `build.sc`:

```scala
import mill._
import mill.scalalib.publish._
import scalalib._

object example extends SbtModule {
  def scalaVersion = "2.13.15"
  override def millSourcePath = os.pwd

  override def ivyDeps = super.ivyDeps() ++ Agg(
    ivy"org.chipsalliance::chisel:6.6.0"
  )

  override def scalacPluginIvyDeps = super.scalacPluginIvyDeps() ++ Agg(
    ivy"org.chipsalliance:::chisel-plugin:6.6.0"
  )
}
```

and the following code (taken from https://www.chisel-lang.org/docs/cookbooks/hierarchy#using-definition-and-instance) to `src/main/scala/Test.scala`:

```scala
import chisel3._
import chisel3.experimental.hierarchy.{Definition, Instance, instantiable, public}

@instantiable
class AddOne(width: Int) extends Module {
  @public val in  = IO(Input(UInt(width.W)))
  @public val out = IO(Output(UInt(width.W)))
  out := in + 1.U
}

class AddTwo(width: Int) extends Module {
  val in  = IO(Input(UInt(width.W)))
  val out = IO(Output(UInt(width.W)))
  val addOneDef = Definition(new AddOne(width))
  val i0 = Instance(addOneDef)
  val i1 = Instance(addOneDef)
  i0.in := in
  i1.in := i0.out
  out   := i1.out
}
```

Compile:

```shell
$ mill example.compile
[39/39] example.compile
[info] compiling 1 Scala source to /home/jiegec/example/out/example/compile.dest/classes ...
[error] /home/jiegec/example/src/main/scala/Test.scala:17:6: value in is not a member of chisel3.experimental.hierarchy.core.Instance[AddOne]
[error]   i0.in := in
[error]      ^
[error] /home/jiegec/example/src/main/scala/Test.scala:18:6: value in is not a member of chisel3.experimental.hierarchy.core.Instance[AddOne]
[error]   i1.in := i0.out
[error]      ^
[error] /home/jiegec/example/src/main/scala/Test.scala:18:15: value out is not a member of chisel3.experimental.hierarchy.core.Instance[AddOne]
[error]   i1.in := i0.out
[error]               ^
[error] /home/jiegec/example/src/main/scala/Test.scala:19:15: value out is not a member of chisel3.experimental.hierarchy.core.Instance[AddOne]
[error]   out   := i1.out
[error]               ^
[error] four errors found
1 targets failed
example.compile Compilation failed
```

**What is the current behavior?**

It fails to compile

**What is the expected behavior?**

It should work

**Please tell us about your environment:**
    - chisel version: `6.6.0`
    - scala version: `2.13.15`
    - mill version: `0.11.12`

jiegec, 2024-11-29:
Updating to mill 0.12.2 fixes the issue. Thanks @unlsycn for guidance.

closed 2024-11-29 by jiegec

B83C, 2025-05-19:
Doesn't work for me on Chisel 6.6.0, scala 2.13.15 and mill 0.12.11. Same error.

B83C, 2025-05-19:
Tried reverting to mill 0.12.2, did not  help here:
```

B83C, 2025-05-19:
Confirmed to work only with sbt and chisel 7.0.0-rc1. Mill doesn't work on all versions

B83C, 2025-05-19:
Found a fix:
```scala
def scalacOptions = T {
  super.scalacOptions() ++ Seq("-Ymacro-annotations")
}
```
As suggested by chatgpt, which is quite accurate, since @public is a macro that expands the members.