chipsalliance/chisel #4567: "Regarding the loss of active reset behavior of Reg or being overridden by RegInit.

linmoIO, 2024-12-18:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
Using the latest chisel-template (corresponding to chisel v6.0.0), emit the following code as System Verilog, and compare the Chisel source code with the resulting System Verilog code.

``` Scala
class Test extends Module {
    val io = IO(new Bundle() {
        val out = Output(Bool())
    })

    val reg = RegInit(false.B)

    reg := true.B
    when(reset.asBool) {
        reg := true.B
    }

    io.out := reg
}

object Test extends App{
    println(ChiselStage.emitSystemVerilog(new Test, firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")))
}
```
"The resulting System Verilog code:
``` Verilog
// Generated by CIRCT firtool-1.62.1
module Test(
  input  clock,
         reset,
  output io_out
);

  reg reg_0;
  always @(posedge clock) begin
    if (reset)
      reg_0 <= 1'h0;
    else
      reg_0 <= 1'h1;
  end // always @(posedge)
  assign io_out = reg_0;
endmodule
```

**What is the current behavior?**
"We observe that in the original code, we explicitly require the value of reg to be true when the reset signal is true. However, in the resulting System Verilog code, the value of the reg signal is false when the reset signal is true.

**What is the expected behavior?**
The reset behavior of the compiled System Verilog code is consistent with the Chisel source code. If this is my misuse, please tell me how to write a reg that is initially false and then remains true. Thank you.

**Please tell us about your environment:**

Chisel version: v6.0.0
OS: Ubuntu 24.04 LTS

**Other Information**

**What is the use case for changing the behavior?**
mentioned before

jackkoenig, 2024-12-19:
The reason for the behavior is that `val reg = RegInit(false.B)` means reset the register to `false` (`0`), this overrides the `when (reset.asBool)`.

I think perhaps the confusion stems from the use of the term "initially false". By "initialization" Chisel means at reset. `RegInit` means "reset the register to the given value. Reading your issue, I am assuming you mean initialization in the sense of a System Verilog initial block. Chisel is intended for writing reusable generators for  synthesizable, synchronous, digital designs. While we recognize that some FPGA tools synthesize initial blocks, not all do, and more importantly, ASIC tools do not. Therefore, Chisel does not support specifying System Verilog initial values for registers (or anything).

pranjal3060, 2024-12-19:
Isn't resest means that it goes to its original state?

linmoIO, 2024-12-19:
Thank you, I understand. It's unfortunate that Chisel does not support initial values like System Verilog.