chipsalliance/chisel #4575: chisel3.experimental.Trace.traceName will report java.lang.NumberFormatException when indexing a Vec chisel type

davine47, 2024-12-24:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
Run the following code:
```
package sandbox

import chisel3.{UInt, _}
import chisel3.util.{Fill, Pipe}
import freechips.rocketchip.diplomacy.{LazyModule, LazyModuleImp, LazyModuleImpLike}
import org.chipsalliance.cde.config.Parameters
import chisel3.experimental.Trace._

class TestAggregate extends Bundle {
  val flag = Bool()
  val value = UInt(3.W)
}

class TestAggregate2 extends Bundle {
  val flag = Bool()
  val value = UInt(3.W)
}

class Hello()(implicit p:Parameters) extends LazyModule {
  lazy val  module = new Impl
  class Impl extends LazyModuleImp(this) {

    val io = IO(new Bundle {
      val in = Input(UInt(8.W))
      val out = Output(UInt(3.W))
    })
    val pipe = Module(new Pipe(UInt(8.W), 40))
    val foo = Wire(Vec(8, new TestAggregate))
    val debug_microOp = Reg(Vec(4, new TestAggregate2))
    foo.foreach(ptr => {
      ptr.flag := 0.B
      ptr.value := 0.U
    })
    val index = foo(0).value
    val tt = debug_microOp(index).value
    pipe.io.enq.bits := io.in
    pipe.io.enq.valid := true.B
    io.out := pipe.io.deq.bits
    traceName(tt) // java.lang.NumberFormatException: For input string: "_tt_T"
  }
}
```
**What is the current behavior?**
Java exceptions
```
[269] java.lang.NumberFormatException: For input string: "_tt_T"
[269] emat java.base/java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
[269] 72at java.base/java.lang.Integer.parseInt(Integer.java:652)
[269] emat java.base/java.lang.Integer.parseInt(Integer.java:770)
[269] 72at scala.collection.StringOps$.toInt$extension(StringOps.scala:908)
[269] emat firrtl.annotations.Target$.$anonfun$toTargetTokens$1(Target.scala:155)
[269] 	at scala.collection.immutable.List.foreach(List.scala:333)
[269] 	at firrtl.annotations.Target$.toTargetTokens(Target.scala:153)
[269] 	at chisel3.internal.NamedComponent.toTarget(Builder.scala:381)
[269] 	at chisel3.internal.NamedComponent.toTarget$(Builder.scala:372)
[269] 	at chisel3.Data.toTarget(Data.scala:286)
[269] 	at chisel3.internal.NamedComponent.toAbsoluteTarget(Builder.scala:389)
[269] 	at chisel3.internal.NamedComponent.toAbsoluteTarget$(Builder.scala:388)
[269] 	at chisel3.Data.toAbsoluteTarget(Data.scala:286)
[269] 	at chisel3.experimental.Trace$$anon$3.toFirrtl(Trace.scala:40)
[269] 72at chisel3.internal.firrtl.Circuit.$anonfun$firrtlAnnotations$1(IR.scala:484)
[269] emat scala.collection.immutable.List.flatMap(List.scala:293)
[269] 	at scala.collection.immutable.List.flatMap(List.scala:79)
[269] 	at chisel3.internal.firrtl.Circuit.firrtlAnnotations(IR.scala:484)
[269] 	at chisel3.stage.phases.Convert.$anonfun$transform$1(Convert.scala:33)
[269] 	at scala.collection.immutable.List.flatMap(List.scala:293)
[269] 	at scala.collection.immutable.List.flatMap(List.scala:79)
[269] 	at chisel3.stage.phases.Convert.transform(Convert.scala:26)
[269] 	at chisel3.stage.phases.Convert.transform(Convert.scala:18)
[269] 	at generators.DiplomacyGenerator$.$anonfun$elaborate$4(DiplomacyGenerator.scala:40)
[269] 	at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
[269] 	at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
[269] 72at scala.collection.immutable.List.foldLeft(List.scala:79)
[269] emat generators.DiplomacyGenerator$.elaborate(DiplomacyGenerator.scala:40)
[269] 	at generators.DiplomacyGenerator$.$anonfun$main$1(DiplomacyGenerator.scala:54)
[269] 	at generators.DiplomacyGenerator$.$anonfun$main$1$adapted(DiplomacyGenerator.scala:54)
[269] 	at mainargs.Invoker$.$anonfun$invoke0$2(Invoker.scala:68)
[269] 	at mainargs.Result.flatMap(Result.scala:13)
[269] 	at mainargs.Result.flatMap$(Result.scala:12)
[269] 	at mainargs.Result$Success.flatMap(Result.scala:23)
[269] 	at mainargs.Invoker$.invoke0(Invoker.scala:67)
[269] 	at mainargs.Invoker$.invoke(Invoker.scala:77)
[269] 	at mainargs.Invoker$.invokeLocal$1(Invoker.scala:89)
[269] 	at mainargs.Invoker$.$anonfun$runMains$2(Invoker.scala:103)
[269] 	at mainargs.Result.flatMap(Result.scala:13)
[269] 	at mainargs.Result.flatMap$(Result.scala:12)
[269] 72at mainargs.Result$Success.flatMap(Result.scala:23)
[269] emat mainargs.Invoker$.groupArgs$1(Invoker.scala:103)
[269] 	at mainargs.Invoker$.runMains(Invoker.scala:108)
[269] 	at mainargs.ParserForMethods.runRaw0(Parser.scala:170)
[269] 	at mainargs.ParserForMethods.runEither(Parser.scala:108)
[269] 	at mainargs.ParserForMethods.runEither(Parser.scala:151)
[269] 	at mainargs.ParserForMethods.runOrExit(Parser.scala:56)
[269] 	at generators.DiplomacyGenerator$.main(DiplomacyGenerator.scala:54)
[269] 	at generators.DiplomacyGenerator.main(DiplomacyGenerator.scala)
[269] 
```

**What is the expected behavior?**
It seems like lacking  assertions or match patterns in function:
`Target.scala -> toTargetTokens`

**Please tell us about your environment:**
    - version: `org.chipsalliance::chisel:6.0.0-RC1`
    - scala: `2.13.10`

light-ly, 2024-12-24:
dontTouch(tt)  also has exception

unlsycn, 2025-02-10:
Does this issue still exist in 6.6.0?

jackkoenig, 2025-02-10:
> Does this issue still exist in 6.6.0?

Surprisingly to me at least, it does still exist in 6.6.0:
```scala
//> using scala "2.13.12"
//> using dep "org.chipsalliance::chisel:6.6.0"
//> using plugin "org.chipsalliance:::chisel-plugin:6.6.0"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
// _root_ disambiguates from package chisel3.util.circt if user imports chisel3.util._
import _root_.circt.stage.ChiselStage

class MyBundle extends Bundle {
  val foo = UInt(8.W)
}

class Foo extends Module {
  val in = IO(Input(Vec(8, new MyBundle)))
  val idx = IO(Input(UInt(3.W)))
  val out = IO(Output(UInt(8.W)))

  val elt = in(idx).foo
  out := elt
  dontTouch(elt)
}

object Main extends App {
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

Thanks for the bug report!

unlsycn, 2025-02-23:
Currently CIRCT does not support dynamically indexed representation in target so that we should first implement https://github.com/llvm/circt/issues/8268

unlsycn, 2025-04-11:
To support this we should implement dynamic indexed `Target`, which I've been willing to do recently.

closed 2025-04-18 by jackkoenig

jackkoenig, 2025-04-18:
I believe I have an even better fix for this in https://github.com/chipsalliance/chisel/pull/4896

It should now error eagerly (at the point of annotating) rather than waiting until emission.

PR #4887 (2025-04-18): Add error message for dynamic index in target
Close #4575 

### Contributor Checklist

- [ ] Did you add Scaladoc to every public function/method?
- [ ] Did you add at least one test demonstrating the PR?
- [ ] Did you delete any extraneous printlns/debugging code?
- [ ] Did you specify the type of improvement?
- [ ] Did you add appropriate documentation in `docs/src`?
- [ ] Did you request a desired merge strategy?
- [ ] Did you add text to be included in the Release Notes for this change?

#### Type of Improvement

- Bugfix

#### Desired Merge Strategy

- Squash: The PR will be squashed and  [... cut]