chipsalliance/chisel #4610: Bug Report: simulation failure

donaldkuck, 2025-01-10:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
use simulate to test code, test code: 
```
class SimtStackTest extends AnyFlatSpec {
  val param = SimtStackParameter(false, false, 32, 40, 2)
  behavior.of("SimtStack")
  it should "push and pop right" in {
    simulate(new SimtStack(param)) { dut =>
      dut.io.clock.step()
      dut.io.reset.poke(true.B)
      dut.io.clock.step()
      dut.io.reset.poke(false.B)
      dut.io.full.expect(false.B)
      dut.io.empty.expect(true.B)
... 
```
**What is the current behavior?**
running failure with message: 
```
SimtStackTest:
SimtStack
*** stack smashing detected ***: terminated
- should push and pop right *** FAILED ***
  svsim.Simulation$UnexpectedEndOfMessages$:
  at svsim.Simulation$UnexpectedEndOfMessages$.<clinit>(Simulation.scala:348)
  at svsim.Simulation$Controller.readNextAvailableMessage(Simulation.scala:140)
  at svsim.Simulation$Controller.$anonfun$completeInFlightCommands$1(Simulation.scala:201)
  at svsim.Simulation$Controller.$anonfun$completeInFlightCommands$1$adapted(Simulation.scala:200)
  at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:619)
  at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:617)
  at scala.collection.AbstractIterable.foreach(Iterable.scala:935)
  at svsim.Simulation$Controller.completeInFlightCommands(Simulation.scala:200)
  at svsim.Simulation$Controller.processNextMessage(Simulation.scala:212)
  at svsim.Simulation$Port.tick(Simulation.scala:406)
  ...
1 targets failed

```
**What is the expected behavior?**
running well on arm macos, failed on x86 ubuntu 24.04

**Please tell us about your environment:**

ubuntu 24.04 x86 failed
macos arm success

**Other Information**

relevent codes: https://github.com/OpenGPGPU/opengpu/blob/main/test/SimtStackTest.scala
**What is the use case for changing the behavior?**

jackkoenig, 2025-01-10:
What version of Chisel is this? Also, is there anything printed above? Any other error messages? I believe that SVSim currently does not handle assertions firing during simulation very elegantly and manifests like this. I got it mostly fixed in https://github.com/chipsalliance/chisel/pull/4386 but that PR still needs to be finished.

donaldkuck, 2025-01-13:
> What version of Chisel is this? Also, is there anything printed above? Any other error messages? I believe that SVSim currently does not handle assertions firing during simulation very elegantly and [...]

Hi, thanks for your reply. I test the latest chisel version ba4e01c3872667faaa37314ccc44aebdb47ef351 . This version is ok.

donaldkuck, 2025-01-13:
> What version of Chisel is this? Also, is there anything printed above? Any other error messages? I believe that SVSim currently does not handle assertions firing during simulation very elegantly and [...]

Thanks for your reply. I finally find that it is not a bug of chisel. It's a problem about my nix config which not works:
```
devShell = pkgs.mkShell.override { stdenv = pkgs.clangStdenv; } {
 ...
}
```

nix config which works:
```
devShell = pkgs.mkShell {
...
}
```

donaldkuck, 2025-01-13:
maybe it is a bug of svsim compilation things?