chipsalliance/chisel #4638: [build] Mill reformat not working after 0.12.5

seldridge, 2025-01-22:
I'm getting a lot of the following errors when running scalafmt after the bump to mill 0.12.5. I don't know if it's a build problem or an upstream mill problem:

```
./mill 'chisel[2.13.15].reformat'
```

I'm using Java 11 (which might matter?):

```
# java --version                   
openjdk 11.0.18 2023-01-17
OpenJDK Runtime Environment GraalVM CE 22.3.1 (build 11.0.18+10-jvmci-22.3-b13)
OpenJDK 64-Bit Server VM GraalVM CE 22.3.1 (build 11.0.18+10-jvmci-22.3-b13, mixed mode)
```

The last error is below. I get a lot of these:
```
[5] error: /Users/schuylereldridge/repos/github.com/chipsalliance/chisel/src/main/scala-2/chisel3/FixedIOModule.scala: java.util.NoSuchElementException: last of empty IndexedSeq
[5] 	at scala.collection.IndexedSeqOps.last(IndexedSeq.scala:110)
[5] 	at scala.collection.IndexedSeqOps.last$(IndexedSeq.scala:105)
[5] 	at scala.meta.tokens.Tokens.last(Tokens.scala:29)
[5] 	at org.scalafmt.internal.Router.$anonfun$getSplitsImpl$69(Router.scala:585)
[5] 	at scala.Option.fold(Option.scala:263)
[5] 	at org.scalafmt.internal.Router.getSplitsImpl(Router.scala:585)
[5] 	at org.scalafmt.internal.Router.getSplits(Router.scala:1957)
[5] 	at org.scalafmt.internal.BestFirstSearch.$anonfun$routes$1(BestFirstSearch.scala:41)
[5] 	at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:619)
[5] 	at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:617)
[5] 	at org.scalafmt.internal.FormatTokens.foreach(FormatTokens.scala:9)
[5] 	at org.scalafmt.internal.BestFirstSearch.<init>(BestFirstSearch.scala:41)
[5] 	at org.scalafmt.internal.BestFirstSearch$.apply(BestFirstSearch.scala:348)
[5] 	at org.scalafmt.Scalafmt$.$anonfun$formatCode$1(Scalafmt.scala:83)
[5] 	at scala.util.Try$.apply(Try.scala:217)
[5] 	at org.scalafmt.Scalafmt$.formatCode(Scalafmt.scala:70)
[5] 	at org.scalafmt.Scalafmt$.format(Scalafmt.scala:52)
[5] 	at org.scalafmt.Scalafmt.format(Scalafmt.scala)
[5] 	at jdk.internal.reflect.GeneratedMethodAccessor50.invoke(Unknown Source)
[5] 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[5] 	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[5] 	at org.scalafmt.dynamic.ScalafmtReflect.$anonfun$tryFormat$1(ScalafmtReflect.scala:114)
[5] 	at scala.util.Try$.apply(Try.scala:217)
[5] 	at org.scalafmt.dynamic.ScalafmtReflect.tryFormat(ScalafmtReflect.scala:110)
[5] 	at org.scalafmt.dynamic.ScalafmtReflectConfig.tryFormat(ScalafmtReflectConfig.scala:91)
[5] 	at org.scalafmt.dynamic.ScalafmtDynamicSession.$anonfun$tryForceFormat$1(ScalafmtDynamicSession.scala:44)
[5] 	at scala.util.Success.flatMap(Try.scala:266)
[5] 	at org.scalafmt.dynamic.ScalafmtDynamicSession.tryForceFormat(ScalafmtDynamicSession.scala:44)
[5] 	at org.scalafmt.dynamic.ScalafmtDynamicSession.tryFormat(ScalafmtDynamicSession.scala:34)
[5] 	at org.scalafmt.dynamic.ScalafmtDynamicSession.format(ScalafmtDynamicSession.scala:19)
[5] 	at org.scalafmt.dynamic.ScalafmtDynamic.format(ScalafmtDynamic.scala:49)
[5] 	at mill.scalalib.scalafmt.ScalafmtWorker.$anonfun$reformatAction$2(ScalafmtWorker.scala:75)
[5] 	at scala.collection.immutable.Vector.foreach(Vector.scala:2125)
[5] 	at mill.scalalib.scalafmt.ScalafmtWorker.reformatAction(ScalafmtWorker.scala:73)
[5] 	at mill.scalalib.scalafmt.ScalafmtWorker.reformat(ScalafmtWorker.scala:22)
[5] 	at mill.scalalib.scalafmt.ScalafmtModule.$anonfun$reformat$2(ScalafmtModule.scala:13)
[5] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.scala:18)
[5] 	at mill.api.Result$.create(Result.scala:25)
[5] 	at mill.scalalib.scalafmt.ScalafmtModule.$anonfun$reformat$1(ScalafmtModule.scala:16)
[5] 	at mill.define.Task$TraverseCtx.evaluate(Task.scala:219)
[5] 	at mill.eval.GroupEvaluator.$anonfun$evaluateGroup$10(GroupEvaluator.scala:267)
[5] 	at mill.eval.GroupEvaluator.$anonfun$evaluateGroup$9(GroupEvaluator.scala:257)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$9(SystemStreams.scala:104)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$8(SystemStreams.scala:103)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$7(SystemStreams.scala:102)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at scala.Console$.withErr(Console.scala:193)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$6(SystemStreams.scala:101)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at scala.Console$.withOut(Console.scala:164)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$5(SystemStreams.scala:100)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at scala.Console$.withIn(Console.scala:227)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$4(SystemStreams.scala:99)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.withStreams(SystemStreams.scala:98)
[5] 	at mill.eval.GroupEvaluator.$anonfun$evaluateGroup$8(GroupEvaluator.scala:257)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.eval.GroupEvaluator.wrap$1(GroupEvaluator.scala:256)
[5] 	at mill.eval.GroupEvaluator.$anonfun$evaluateGroup$3(GroupEvaluator.scala:266)
[5] 	at mill.eval.GroupEvaluator.$anonfun$evaluateGroup$3$adapted(GroupEvaluator.scala:215)
[5] 	at scala.collection.immutable.Vector.foreach(Vector.scala:2125)
[5] 	at mill.eval.GroupEvaluator.computeAll$1(GroupEvaluator.scala:215)
[5] 	at mill.eval.GroupEvaluator.evaluateGroup(GroupEvaluator.scala:287)
[5] 	at mill.eval.GroupEvaluator.$anonfun$evaluateGroupCached$1(GroupEvaluator.scala:156)
[5] 	at mill.api.Logger.withPrompt(Logger.scala:83)
[5] 	at mill.api.Logger.withPrompt$(Logger.scala:81)
[5] 	at mill.util.PrefixLogger.withPrompt(PrefixLogger.scala:7)
[5] 	at mill.eval.GroupEvaluator.evaluateGroupCached(GroupEvaluator.scala:58)
[5] 	at mill.eval.GroupEvaluator.evaluateGroupCached$(GroupEvaluator.scala:44)
[5] 	at mill.eval.EvaluatorImpl.evaluateGroupCached(EvaluatorImpl.scala:17)
[5] 	at mill.eval.EvaluatorCore.$anonfun$evaluate0$6(EvaluatorCore.scala:175)
[5] 	at scala.concurrent.impl.Promise$Transformation.run(Promise.scala:467)
[5] 	at mill.eval.ExecutionContexts$ThreadPool$$anon$1.$anonfun$run$3(ExecutionContexts.scala:69)
[5] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.scala:18)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$9(SystemStreams.scala:104)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$8(SystemStreams.scala:103)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$7(SystemStreams.scala:102)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at scala.Console$.withErr(Console.scala:193)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$6(SystemStreams.scala:101)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at scala.Console$.withOut(Console.scala:164)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$5(SystemStreams.scala:100)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at scala.Console$.withIn(Console.scala:227)
[5] 	at mill.api.SystemStreams$.$anonfun$withStreams$4(SystemStreams.scala:99)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.api.SystemStreams$.withStreams(SystemStreams.scala:98)
[5] 	at mill.eval.ExecutionContexts$ThreadPool$$anon$1.$anonfun$run$2(ExecutionContexts.scala:69)
[5] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.scala:18)
[5] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
[5] 	at mill.eval.ExecutionContexts$ThreadPool$$anon$1.run(ExecutionContexts.scala:68)
[5] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
[5] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[5] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
[5] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
[5] 	at java.base/java.lang.Thread.run(Thread.java:829)
```

seldridge, 2025-01-23:
I'm seeing this fail in CI now: https://github.com/chipsalliance/chisel/actions/runs/12920226523/job/36032184535?pr=4628

@sequencer: Any idea what is going on here?

Emin017, 2025-01-23:
It works with the newer version of Scalafmt, (I successfully format them with version 3.8.5)

seldridge, 2025-01-23:
Ah, the problem is that the `.scalafmt.conf` file was not updated. I'm not sure how you are doing formatting with a newer version of Scalafmt given that the config file is using 2.6.4. I'll see about fixing this.

Emin017, 2025-01-23:
> Ah, the problem is that the `.scalafmt.conf` file was not updated. I'm not sure how you are doing formatting with a newer version of Scalafmt given that the config file is using 2.6.4. I'll see about fixing this.

I just changed the version in .scalafmt.conf to 3.8.5, and made some migrations to the new version of api. Although the formating can be completed normally, the code style will still change a lot.

closed 2025-01-23 by jackkoenig

jackkoenig, 2025-01-23:
@Emin017 I merged the upgrade in #4566

PR #4566 (2025-01-23): Upgrade Scalafmt to 3.8.5 and reformat
As you can see, there's a bit of a diff. I've done the best I can to minimize it but there's some diff.

It seems to try to keep argument lists on one line, which I like, but it does it even if an early argument list was split between multiple lines which is a bit weird.

It also likes keeping `case` on the same line as the `{` which I like, although occasionally that leads to longish lines. Anyway, we can try to massage this further or just go with it. I think it's fine but am curious what others think.

We will need to bump Scalafmt soon in order to sup [... cut]