chipsalliance/chisel #4666: Hash Mismatch with chisel publishLocal across different Machines

Emin017, 2025-02-06:
I'm trying to build `gcd.vcs.rtl` on branch [mill-bump](https://github.com/chipsalliance/chisel-nix/tree/mill-bump)
steps: 
```shell
cd templates/chisel && nix build .#gcd.vcs.rtl
```

## Descriptions
I'm encountering a hash mismatch issue when using `mill -i unipublish.publishLocal` on two different x86_64-linux NixOS machines. It seems that the `outputHash` specified in `publishMillModule` will changes on different machines.

I followed the method described [here](https://github.com/chipsalliance/chisel-nix/pull/62#issuecomment-2628953969) to compare the differences between outputs (`.ivy2/local`) on two machines. And I've narrowed down the problem to `.ivy2/local/org.chipsalliance/chisel_2.13/0.0.0+0-no-vcs-SNAPSHOT/docs/chisel_2.13-javadoc.jar`.  I unpacked it using `unzip` and got the following diff:
<details>
<summary>Click to expand</summary>

```diff
diff -bur result2-unzip/chisel3/connectable/Connectable.html result3-unzip/chisel3/connectable/Connectable.html
--- result2-unzip/chisel3/connectable/Connectable.html	2022-11-30 12:13:00
+++ result3-unzip/chisel3/connectable/Connectable.html	2022-11-30 12:13:00
@@ -21,7 +21,7 @@
 }
   subgraph clusterIncoming {
   style = "invis"
-  node2 [label="48 classes/traits", id="graph1_2", class="implicit-incoming()", tooltip="SRAMInterface[T], MemoryReadWritePort[T], MemoryWritePort[T], MemoryReadPort[T], MixedVec[T], DecoupledIO[T], ReadyValidIO[T], PRNGIO, PipeIO, CIRCTSRAMInterface, CIRCTSRAMWritePort, CIRCTSRAMReadWritePort, CIRCTSRAMReadPort, AutoBundleFromVerilog, DecodeBundle, SparseVec[A], ArbiterIO[T], IrrevocableIO[T], QueueIO[T], Valid[T], PlusArgsRetBundle[T], Property[T], HWTuple10[A, B, C, D, E, F, G, H, I, J], HWTuple9[A, B, C, D, E, F, G, H, I], HWTuple8[A, B, C, D, E, F, G, H], HWTuple7[A, B, C, D, E, F, G], HWTuple6[A, B, C, D, E, F], HWTuple5[A, B, C, D, E], HWTuple4[A, B, C, D], HWTuple3[A, B, C], HWTuple2[A, B], OpaqueType, Analog, Vec[T], UInt, SInt, ResetType, Reset, Record, EnumType, Element, Data, Clock, Bundle, Bool, Bits, AsyncReset, Aggregate"] ;
+  node2 [label="48 classes/traits", id="graph1_2", class="implicit-incoming()", tooltip="SRAMInterface[T], MemoryReadWritePort[T], MemoryWritePort[T], MemoryReadPort[T], ArbiterIO[T], Valid[T], MixedVec[T], ReadyValidIO[T], PRNGIO, IrrevocableIO[T], SparseVec[A], QueueIO[T], PipeIO, DecoupledIO[T], AutoBundleFromVerilog, DecodeBundle, CIRCTSRAMInterface, CIRCTSRAMWritePort, CIRCTSRAMReadWritePort, CIRCTSRAMReadPort, PlusArgsRetBundle[T], Property[T], HWTuple10[A, B, C, D, E, F, G, H, I, J], HWTuple9[A, B, C, D, E, F, G, H, I], HWTuple8[A, B, C, D, E, F, G, H], HWTuple7[A, B, C, D, E, F, G], HWTuple6[A, B, C, D, E, F], HWTuple5[A, B, C, D, E], HWTuple4[A, B, C, D], HWTuple3[A, B, C], HWTuple2[A, B], OpaqueType, Analog, Vec[T], UInt, SInt, ResetType, Reset, Record, EnumType, Element, Data, Clock, Bundle, Bool, Bits, AsyncReset, Aggregate"] ;
   
                }
   node2 -> node0[label="implicitly", id="2_0", minlen="2", ltail="clusterIncoming", lhead="clusterThis", constraint="false", class="implicit-incoming", tooltip="48 classes/traits can be implicitly converted to Connectable[T]"]
diff -bur result2-unzip/chisel3/experimental/BaseModule$$BaseModuleExtensions.html result3-unzip/chisel3/experimental/BaseModule$$BaseModuleExtensions.html
--- result2-unzip/chisel3/experimental/BaseModule$$BaseModuleExtensions.html	2022-11-30 12:13:00
+++ result3-unzip/chisel3/experimental/BaseModule$$BaseModuleExtensions.html	2022-11-30 12:13:00
@@ -17,7 +17,7 @@
 }
   subgraph clusterIncoming {
   style = "invis"
-  node2 [label="39 classes/traits", id="graph1_2", class="implicit-incoming()", tooltip="FixedIOExtModule[A], FixedIORawModule[A], FixedIOBaseModule[A], Stub, interface.Interface.Wrapper.Module[B], interface.Interface.Wrapper.BlackBox, BasicTester, SRAMBlackbox, SRAMDescription, HasExtModulePath, HasExtModuleInline, HasExtModuleResource, PRNG, LFSR, MaxPeriodGaloisLFSR, GaloisLFSR, MaxPeriodFibonacciLFSR, FibonacciLFSR, Pipe[T], CIRCTSRAM[T], AutoBlackBox, HasBlackBoxPath, HasBlackBoxInline, HasBlackBoxResource, Arbiter[T], RRArbiter[T], LockingArbiter[T], LockingRRArbiter[T], LockingArbiterLike[T], Queue[T], Class, IntrinsicModule, ExtModule, BaseModule, RequireSyncReset, RequireAsyncReset, RawModule, Module, BlackBox"] ;
+  node2 [label="39 classes/traits", id="graph1_2", class="implicit-incoming()", tooltip="FixedIOExtModule[A], FixedIORawModule[A], FixedIOBaseModule[A], Stub, interface.Interface.Wrapper.Module[B], interface.Interface.Wrapper.BlackBox, BasicTester, SRAMBlackbox, SRAMDescription, Arbiter[T], RRArbiter[T], LockingArbiter[T], LockingRRArbiter[T], LockingArbiterLike[T], MaxPeriodGaloisLFSR, GaloisLFSR, MaxPeriodFibonacciLFSR, FibonacciLFSR, PRNG, LFSR, HasExtModulePath, HasExtModuleInline, HasExtModuleResource, HasBlackBoxPath, HasBlackBoxInline, HasBlackBoxResource, Queue[T], Pipe[T], AutoBlackBox, CIRCTSRAM[T], Class, IntrinsicModule, ExtModule, BaseModule, RequireSyncReset, RequireAsyncReset, RawModule, Module, BlackBox"] ;
   
                }
   node2 -> node0[label="implicitly", id="2_0", minlen="2", ltail="clusterIncoming", lhead="clusterThis", constraint="false", class="implicit-incoming", tooltip="39 classes/traits can be implicitly converted to BaseModuleExtensions[T]"]
diff -bur result2-unzip/chisel3/experimental/hierarchy/core/IsInstantiable$$IsInstantiableExtensions.html result3-unzip/chisel3/experimental/hierarchy/core/IsInstantiable$$IsInstantiableExtensions.html
--- result2-unzip/chisel3/experimental/hierarchy/core/IsInstantiable$$IsInstantiableExtensions.html	2022-11-30 12:13:00
+++ result3-unzip/chisel3/experimental/hierarchy/core/IsInstantiable$$IsInstantiableExtensions.html	2022-11-30 12:13:00
@@ -17,7 +17,7 @@
 }
   subgraph clusterIncoming {
   style = "invis"
-  node2 [label="40 classes/traits", id="graph1_2", class="implicit-incoming()", tooltip="FixedIOExtModule[A], FixedIORawModule[A], FixedIOBaseModule[A], Stub, interface.Interface.Wrapper.Module[B], interface.Interface.Wrapper.BlackBox, BasicTester, SRAMBlackbox, SRAMDescription, HasExtModulePath, HasExtModuleInline, HasExtModuleResource, PRNG, LFSR, MaxPeriodGaloisLFSR, GaloisLFSR, MaxPeriodFibonacciLFSR, FibonacciLFSR, Pipe[T], CIRCTSRAM[T], AutoBlackBox, HasBlackBoxPath, HasBlackBoxInline, HasBlackBoxResource, Arbiter[T], RRArbiter[T], LockingArbiter[T], LockingRRArbiter[T], LockingArbiterLike[T], Queue[T], Class, IsInstantiable, IntrinsicModule, ExtModule, BaseModule, RequireSyncReset, RequireAsyncReset, RawModule, Module, BlackBox"] ;
+  node2 [label="40 classes/traits", id="graph1_2", class="implicit-incoming()", tooltip="FixedIOExtModule[A], FixedIORawModule[A], FixedIOBaseModule[A], Stub, interface.Interface.Wrapper.Module[B], interface.Interface.Wrapper.BlackBox, BasicTester, SRAMBlackbox, SRAMDescription, Arbiter[T], RRArbiter[T], LockingArbiter[T], LockingRRArbiter[T], LockingArbiterLike[T], MaxPeriodGaloisLFSR, GaloisLFSR, MaxPeriodFibonacciLFSR, FibonacciLFSR, PRNG, LFSR, HasExtModulePath, HasExtModuleInline, HasExtModuleResource, HasBlackBoxPath, HasBlackBoxInline, HasBlackBoxResource, Queue[T], Pipe[T], AutoBlackBox, CIRCTSRAM[T], Class, IsInstantiable, IntrinsicModule, ExtModule, BaseModule, RequireSyncReset, RequireAsyncReset, RawModule, Module, BlackBox"] ;
   
                }
   node2 -> node0[label="implicitly", id="2_0", minlen="2", ltail="clusterIncoming", lhead="clusterThis", constraint="false", class="implicit-incoming", tooltip="40 classes/traits can be implicitly converted to IsInstantiableExtensions[T]"]
diff -bur result2-unzip/chisel3/simulator/index.html result3-unzip/chisel3/simulator/index.html
--- result2-unzip/chisel3/simulator/index.html	2022-11-30 12:13:00
+++ result3-unzip/chisel3/simulator/index.html	2022-11-30 12:13:00
@@ -19,18 +19,18 @@
   
   node0 [URL="package$$AnySimulatedModule.html#inheritance-diagram-container", label="AnySimulatedModule", id="graph1_0", class="default", tooltip="chisel3.simulator.AnySimulatedModule"] ;
   node1 [URL="package$$SimulatedModule.html#inheritance-diagram-container", label="SimulatedModule[T]", id="graph1_1", class="default", tooltip="chisel3.simulator.SimulatedModule"] ;
-  node2 [URL="EphemeralSimulator$.html#inheritance-diagram-container", label="EphemeralSimulator", id="graph1_2", class="default", tooltip="chisel3.simulator.EphemeralSimulator"] ;
[... cut ...] 67da1b, but I'm aware that this modified version makes some inappropriate assumptions (because I assumed that the dependencies only need to publishLocal the chisel dependency, so that it can remove `chisel_2.13-javadoc.jar`), but does this `chisel_2.13-javadoc.jar` really need to be passed to the RTL build? I unpacked it with `unzip` and found that it's full of HTML pages, which doesn't seem necessary in the development process. Can we just remove it? Or is this kind of hash mismatch acceptable?

sequencer, 2025-02-06:
I transfer this issue to chisel, the non-reproducibility should be fixed in upstream.

Avimitin, 2025-02-06:
Date time is handled by add-determinism and should not be a problem. I think the current issue is that the document is generated with random seeds or random sort strategy, which create inconsistency in output HTML files.

jackkoenig, 2025-02-06:
There's a good chance this is an issue in ScalaDoc itself and, if so, there's not a lot we can do here. We should check if there are options or anything but this smells like a problem in ScalaDoc to me.

Emin017, 2025-02-07:
> can you try to add `--date` to `javaDoc` options in Chisel? I suspect it should work.
> 
> ```
> override def javadocOptions = Task(super.javadocOptions() ++ Seq("--date", "1970-01-01T00:00:00.000Z"))
> ```

It doesn't work.

sequencer, 2025-02-07:
Try diff the out dir to see if there are any non-reproducibility in mill. If that’s stable, we have to blame to the ScalaDoc.

Emin017, 2025-02-07:
Thanks to @Anillc 's help, we've found the relevant code implementation for generating nodes in the Scala compiler:

https://github.com/scala/scala/blob/2.13.x/src/scaladoc/scala/tools/nsc/doc/model/diagram/DiagramFactory.scala#L108

https://github.com/scala/scala/blob/2.13.x/src/scaladoc/scala/tools/nsc/doc/model/diagram/Diagram.scala#L124

It seems that the traversal algorithm they are using does not guarantee a consistent order of nodes.

sequencer, 2025-02-07:
🫠 okay. This issue should be transfer to Scala. Let’s discuss at Scala issue page and link to here

closed 2025-02-07 by sequencer

Emin017, 2025-02-07:
> Try diff the out dir to see if there are any non-reproducibility in mill. If that’s stable, we have to blame to the ScalaDoc.

Mill uses the scalaDoc part of Scala compiler: https://github.com/com-lihaoyi/mill/blob/main/scalalib/src/mill/scalalib/Lib.scala#L229