chipsalliance/chisel #4668: Long logic expression operation conversion error.

fanliyong007, 2025-02-07:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
//> using scala "2.13.12"
//> using dep "org.chipsalliance::chisel:6.5.0"
//> using plugin "org.chipsalliance:::chisel-plugin:6.5.0"
//> using options  "-language:reflectiveCalls","-deprecation", "-feature", "-Xcheckinit", "Xcheckinit", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"
import chisel3._
import chisel3.util.Cat
import circt.stage.ChiselStage

class Rule110 extends Module with ImplicitClock {
  val clk = IO(Input(Clock()))
  val load = IO(Input(Bool()))
  val data= IO(Input(UInt(512.W)))
  val q = IO(Output(UInt(512.W)))
  override protected def implicitClock = clk
  val regQ = Reg(UInt(512.W))
  when(load){
    regQ:=data
  }.otherwise {
    val shifted = Cat(regQ(510, 0), 0.U)
    regQ := ((regQ ^ shifted) & regQ(511, 1)) | ((regQ | shifted) & ( ~ regQ(511, 1)))// bug in here
//    (((q[511:0] ^ {q[510:0], 1'b0}) & q[511:1]) | ((q[511:0] | {q[510:0], 1'b0}) & (~q[511:1])));   this is the correct verilog code.

  }
  q:=regQ
}
**What is the current behavior?**
This generates:
// Generated by CIRCT firtool-1.62.0
module Rule110(
  input          clock,
                 reset,
                 clk,
                 load,
  input  [511:0] data,
  output [511:0] q
);

  reg [511:0] regQ;
  always @(posedge clk) begin
    if (load)
      regQ <= data;
    else begin
      automatic logic [510:0] _GEN = {regQ[509:0], 1'h0};
      regQ <=
        {1'h0,                                                                                                                          //wrong is here
         (regQ[510:0] ^ _GEN) & regQ[511:1] | (regQ[510:0] | _GEN) & ~(regQ[511:1])};
    end
  end // always @(posedge)
  assign q = regQ;
endmodule
**What is the expected behavior?**
  {1'h0,  (regQ[510:0] ^ _GEN) & regQ[511:1] | (regQ[510:0] | _GEN) & ~(regQ[511:1])};  //wrong verilog code
(((q[511:0] ^ {q[510:0], 1'b0}) & q[511:1]) | ((q[511:0] | {q[510:0], 1'b0}) & (~q[511:1])));   //this is the correct verilog code.

**Please tell us about your environment:**

**Other Information**

**What is the use case for changing the behavior?**

fanliyong007, 2025-02-07:
when regQ is :73c7461f09a33e269beef89be26f8e26e9be26f89bef8ee3877c0bc4df137c4d0000000000000000000000000000000000000000000000000000000000000000
the next should be:d64dce311be7626fbe3b89be26f89a6fbbe26f89be389ba68dc41e4df137c4df0000000000000000000000000000000000000000000000000000000000000000
but in my code it will be:564dce311be7626fbe3b89be26f89a6fbbe26f89be389ba68dc41e4df137c4df0000000000000000000000000000000000000000000000000000000000000000

jackkoenig, 2025-02-07:
Zooming in on just half of the expression:
```scala
(regQ ^ shifted) & regQ(511, 1)
```
`reqQ` and `shifted` are each 512 bits, so that's fine, but `regQ(511, 1)` is 511 bits. The semantics of Chisel are to zero-extend it to 512 bits, so you effectively get:
```
(regQ ^ shifted) & Cat(0.U, regQ(511, 1))
```
Which can then be legally transformed in compilation to (AND of anything with `0` is `0` so we can effectively factor out the `0`)
```
Cat(0.U, (regQ(510, 0) ^ shifted(510, 0)) & regQ(511, 1))
```

I'm not sure what the semantics of `&` in Verilog is when the widths are not the same, maybe it's sign extending? But these are `UInts` so in Chisel it's a zero-extension always.

fanliyong007, 2025-02-08:
Thank you for your explanation.

closed 2025-02-08 by fanliyong007