chipsalliance/chisel #4733: literal.asUInt(width) always returns a Bool

tymcauley, 2025-02-26:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

```scala
//> using scala 2.13.15
//> using repository https://s01.oss.sonatype.org/content/repositories/snapshots
//> using dep org.chipsalliance::chisel:7.0.0-M2+432-486b55dc-SNAPSHOT
//> using plugin org.chipsalliance:::chisel-plugin:7.0.0-M2+432-486b55dc-SNAPSHOT
//> using options -unchecked -deprecation -language:reflectiveCalls -feature -Xcheckinit -Xfatal-warnings -Ywarn-dead-code -Ywarn-unused -Ymacro-annotations

import chisel3._

class Foo(x: BigInt, width: Int) extends Module {
  val out1 = IO(Output(UInt(width.W)))
  val out2 = IO(Output(UInt(width.W)))
  val out3 = IO(Output(UInt(width.W)))
  out1 := x.asUInt(width)
  out2 := x.asUInt
  out3 := x.U(width.W)
}

object Main extends App {
  import _root_.circt.stage.ChiselStage
  val width = 8
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo((1 << width) - 1, width),
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info", "-default-layer-specialization=enable")
    )
  )
}
```

Running this with `scala-cli` generates this result:

```verilog
// Generated by CIRCT firtool-1.107.0
module Foo(
  input        clock,
               reset,
  output [7:0] out1,
               out2,
               out3
);

  assign out1 = 8'h0;
  assign out2 = 8'hFF;
  assign out3 = 8'hFF;
endmodule
```

**What is the current behavior?**

Running `x.asUInt(width)` always returns a `Bool` (the same happens if you change `x` from a `BigInt` to an `Int`). If you change the `out1 := x.asUInt(width)` line above to use the `:<=` operator, you'll get this error:

```
[error] ./chisel-example.scala:13:3
[error] Cannot prove that chisel3.UInt =:= chisel3.Bool.
[error]   out1 :<= x.asUInt(width)
[error]   ^^^^^^^^^^^^^^^^^^^^^^^^
```

**What is the expected behavior?**

`x.asUInt(width)` should return a `UInt` of the specified width. I would expect the resulting Verilog to say `assign out1 = 8'hFF;`, like `out2` and `out3`.

**Please tell us about your environment:**

- version: `7.0.0-M2+432-486b55dc-SNAPSHOT`
- OS: macOS 15.3.1

**Other Information**

**What is the use case for changing the behavior?**

This should match the behavior of `x.U(width.W)`.

tymcauley, 2025-02-26:
Ha, I see the issue was me not using `x.asUInt(width.W)`. Perhaps this should give an error/warning rather than silently allowing the user to pass an `Int` for the `Width`?

jackkoenig, 2025-02-26:
It's really hard to catch this sort of thing--we did create a macro for catching a similar case, e.g. `4.U(8)`: https://github.com/chipsalliance/chisel/blob/f2f3adefb29495d0eca52356e8f6d00d97d1465c/macros/src/main/scala/chisel3/internal/sourceinfo/SourceInfoTransform.scala#L357

We could try expanding that macro to catch this. Another option might be to define `asUInt(x: Int)` and have it always macro expand into an error. That might be simpler than the existing macro at least.

tymcauley, 2025-02-26:
Yup, that makes sense. On one hand, this feels like a serious foot-gun, and is almost never intended (`asUInt` with an immediate one-bit extract). On the other hand, that's the current API, and some users might be intentionally using it right now. I would feel fine with either solution: making `.asUInt(x: Int)` a warning with the suggestion to use `.asUInt.extract(x: Int)` instead, or making `.asUInt(x: Int)` a hard error. But leaving this as-is seems like a bad idea.

jackkoenig, 2025-03-03:
Lol at this comment:

https://github.com/chipsalliance/chisel/blob/2ec31591e6f40387f6bf74494f0b18ca712b0c02/core/src/main/scala/chisel3/package.scala#L34-L36

It seems once upon a time we had found a way to stop people accidentally chaining the apply, but I think in one of the moves to Scala 2.13 style we dropped the trailing `()` without realizing the utility it was providing. So another option instead of removing `apply` for bit extraction is to require trailing `()` on `.asUInt()` and `.U()`. That has a big negative of being inconsistent with other 0-arity methods though and inconsistent with Scala-style that `()` on a 0-arity method implies mutation.

mwachs5, 2025-03-03:
I'd argue that `asUInt()` *does* imply mutation. Similarly .asBools(). You can't put an annotation on the result of asUInt or .asBools and have it refer to the thing before the .

jackkoenig, 2025-03-03:
FWIW, the case we're talking about here is calling `.asUInt` on a literal, not normal hardware. A related idea is to just get rid of `.asUInt` and `.asSInt` on literals.

> You can't put an annotation on the result of asUInt or .asBools and have it refer to the thing before the .

A new object with different identity still doesn't mean mutation though, calling `.asList` on a `Set` returns something very different than the original `Set`, but it's not mutation.