chipsalliance/chisel #4785: LM_LICENSE_FILE environment variable required for ChiselSim using VCS

tymcauley, 2025-03-11:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

Run any ChiselSim test using `chisel3.simulator.HasSimulator.simulators.vcs` as the simulator backend, but don't define the `LM_LICENSE_FILE` environment variable.

**What is the current behavior?**

This throws an exception from here: https://github.com/chipsalliance/chisel/blob/a7effb7da1db69adeacb1853a2e74369a4e48609/src/main/scala/chisel3/simulator/HasSimulator.scala#L64-L68

The code searching for this environment variable comes from here: https://github.com/chipsalliance/chisel/blob/a7effb7da1db69adeacb1853a2e74369a4e48609/svsim/src/main/scala/vcs/Backend.scala#L73

**What is the expected behavior?**

ChiselSim shouldn't require this particular environment variable for VCS simulations.

**Please tell us about your environment:**
version: `7.0.0-M2+529-5f2e70c3-SNAPSHOT`
OS: RHEL 8

**Other Information**

It appears that if `SNPSLMD_LICENSE_FILE` is defined, VCS will use that and ignore `LM_LICENSE_FILE`.

That being said, does ChiselSim (or rather svsim) need to be prescriptive about this? What's the need for svsim to grab this environment variable and re-export it during the simulation?

This was discussed here: https://github.com/chipsalliance/chisel/pull/3121#discussion_r1144281412

**What is the use case for changing the behavior?**

Enable more uses to use the VCS backend for ChiselSim.

seldridge, 2025-03-11:
Yeah, this is wrong. It should be either.

VCS docs say that you must set either `LM_LICSENSE_FILE` or `SNPSLMD_LICENSE_FILE` with the latter having higher priority.

seldridge, 2025-03-11:
Looking at this more, I'm actually not sure if checking _any_ of this is necessary/required. `svsim` seems to be capturing environment variables and then re-emitting them into the `Makefile` as part of the invocation of specific tools. None of that should be strictly necessary.

The only benefit I can see here is that it is advantageous to check that a tool is available before letting a `Makefile` target fail with an unexpected error. However, the actual _capturing_ of the environment shouldn't be necessary.

Much of this was done to try to make this work with the SiFive-internal VCS flow. (Though the internal implementation uses a completely custom VCS Backend... and everything I've been doing with ChiselSim is building towards beefing up the open source one and then migrating the internal one onto this.)

tymcauley, 2025-03-12:
> Looking at this more, I'm actually not sure if checking _any_ of this is necessary/required. `svsim` seems to be capturing environment variables and then re-emitting them into the `Makefile` as part [...]

Ya, agreed, I don't really understand why the environment needs to be captured and then re-exported. Seems like `svsim` could just be agnostic to these environment variables.

seldridge, 2025-03-12:
I realized why it works this way...

The problem that this is trying to solve is that you want to be able to _rerun_ any test. There is no guarantee that when the test runs through svsim, that you will be in the same environment as when you do the rerun.

We have a flow that is roughly like this:

```
make runScalaTest
```

As part of the above command, this automatically loads an environment which has VCS.

However, if a test fails, you want to be able to go into the build directory and do a `make replay` to debug the failure. However, if the environment to run VCS is not captured, then the user won't be able to easily just run `make replay`. They'll have to setup VCS.

There's some utility in this. However, I think this should just be "capture the environment variables important to VCS based on the VCS manual" and we're diverging from that by ignoring `SNPSLMD_LICENSE_FILE`.

PR #4792 (2025-03-13): [svsim] Use SNPS-specific license in VCS
Fix the logic for grabbing environment variables for the replay Makefile when running VCS tests.  The VCS manual says that you grab the Synopsys-specific license file (SNPSLMD_LICENSE_FILE) first and only use the generic license file (LM_LICENSE_FILE) if that fails.  The existing code was wrong in that it only grabbed the latter.  There is no requirement that the latter is available and that is only a fallback if the former doesn't exist.

Also, add some lightweight infrastructure to cleanup the representation of VCS license files.

I tested this intern [... cut]
commit df891050 (2025-03-13): [svsim] Use SNPS-specific license in VCS (#4792)

Fix the logic for grabbing environment variables for the replay Makefile
when running VCS tests.  The VCS manual says that you grab the
Synopsys-specific license file (SNPSLMD_LICENSE_FILE) first and only use
the generic license file (LM_LICENSE_FILE) if that fails.  The existing
code was wrong in that it only grabbed the latter.  There is no
requirement that the latter is available and that is only a fallback if
the former doesn't exist.

Also, add some lightweight infrastructure to cleanup the representation of
VCS license files.

I tested th [... cut]
closed 2025-03-13 by seldridge