chipsalliance/chisel #4791: Bug in simulation verification

tobias1012, 2025-03-13:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
Any Test on windows 10 should recreate this behaviour

**What is the current behavior?**
The simulator (somewhere) tries to create files that are not valid unicode.
https://github.com/chipsalliance/chisel/blob/d94ae23c32429b902ce11e073d1c2ed09862e725/core/src/main/scala/chisel3/layers/Layers.scala#L14-L18
does not seem to have any effect on the created directories even though the firtool tries to create an output related to these layers. Note the code shows lowercase while the output is trying to create uppercase.

**What is the expected behavior?**
folders for the layers are created correctly

**Please tell us about your environment:**
version: master snapshot
OS: Windows 10

**Other Information**
Error:
`[info] <stdin>:0:0: error: cannot create output directory "C:\Users\...\AppData\Local\Temp\chiselsim-2025-03-13T11-39-25.1344942003004127992777573105\support-artifacts\Verification?ver": Unicode-symbol does not exist in the multibyte symbol-table.
`
Generated SystemVerilog for simulating
``` bash
[info]   "sv.verbatim"() <{format_string = "`include \22Verification\\layers-MyModule-Verification.sv\22\0A`ifndef layers_MyModule_Verification_Cover\0A`define layers_MyModule_Verification_Cover", symbols = []}> {output_file = #hw.output_file<"Verification\FFver\\layers-MyModule-Verification-Cover.sv", excludeFromFileList>} : () -> ()
[info]   "sv.verbatim"() <{format_string = "`include \22Verification\\layers-MyModule-Verification.sv\22\0A`ifndef layers_MyModule_Verification_Assume\0A`define layers_MyModule_Verification_Assume", symbols = []}> {output_file = #hw.output_file<"Verification\FFsume\\layers-MyModule-Verification-Assume.sv", excludeFromFileList>} : () -> ()
[info]   "sv.verbatim"() <{format_string = "`include \22Verification\\layers-MyModule-Verification.sv\22\0A`ifndef layers_MyModule_Verification_Assert\0A`define layers_MyModule_Verification_Assert", symbols = []}> {output_file = #hw.output_file<"Verification\FFsert\\layers-MyModule-Verification-Assert.sv", excludeFromFileList>} : () -> ()
```
**What is the use case for changing the behavior?**
Support on windows

tobias1012, 2025-03-13:
My guess is this is a bug in the interaction of chisel and firtool, however I have not been able to find any code that could generate the files wrong

seldridge, 2025-03-13:
I agree I suspect this is something in the Chisel or Chisel/firtool interaction as we do run all CIRCT tests on Windows.

tobias1012, 2025-03-13:
The interesting part here is there seems to be no issues when using chisel.stage.emitSystemVerilog and manually adding layers