chipsalliance/chisel #4809: Mux Tree Generation

escou64, 2025-03-18:
Hi,

I need to constraint the way multiplexer trees are generated.
Currently, with Chisel 6.5.0, the following code:
```scala
class MuxTree(nBit: Int) extends Module {
  val io = IO(new Bundle {
    val i_slct = Input(UInt(2.W))
    val i_in = Input(Vec(4, UInt(nBit.W)))   

    val o_out = Output(UInt(nBit.W))  
  })  

  io.o_out := io.i_in(io.i_slct)
}
```

leads to the following SystemVerilog:
```sv
// Generated by CIRCT firtool-1.62.0
module MuxTree(
  input        clock,
               reset,
  input  [1:0] io_i_slct,
  input  [3:0] io_i_in_0,
               io_i_in_1,
               io_i_in_2,
               io_i_in_3,
  output [3:0] io_o_out
);

  wire [3:0][3:0] _GEN = {{io_i_in_3}, {io_i_in_2}, {io_i_in_1}, {io_i_in_0}};
  assign io_o_out = _GEN[io_i_slct];
endmodule
```
In my case, I want the different multiplexers to be explicitly unrolled with intermediate signals, with the output of each 2:1 multiplexer. For example here:
```sv
// Manually written
module MuxTree(
  input        clock,
               reset,
  input  [1:0] io_i_slct,
  input  [3:0] io_i_in_0,
               io_i_in_1,
               io_i_in_2,
               io_i_in_3,
  output [3:0] io_o_out
);

wire [3:0] _GEN0_0;
wire [3:0] _GEN0_1;
wire [3:0] _GEN1_0;

assign _GEN0_0 = (io_i_slct[0] == 1'b1) ? io_i_in_1 : io_i_in_0;
assign _GEN0_1 = (io_i_slct[0] == 1'b1) ? io_i_in_3 : io_i_in_2;
assign _GEN1_0 = (io_i_slct[1] == 1'b1) ? _GEN0_1 : _GEN0_0;

assign io_o_out = _GEN1_0;
endmodule
```

Is it possible to force this kind of behavior in Chisel ?
Thanks for your support.

jackkoenig, 2025-03-18:
Firtool has some lower-options, e.g. if you pass `-lowering-options=disallowPackedArrays` it will emit as a `casez`:
```verilog
module MuxTree(
  input        clock,
               reset,
  input  [2:0] io_i_slct,
  input  [7:0] io_i_in_0,
               io_i_in_1,
               io_i_in_2,
               io_i_in_3,
  output [7:0] io_o_out
);

  reg [7:0] casez_tmp;
  always_comb begin
    casez (io_i_slct[1:0])
      2'b00:
        casez_tmp = io_i_in_0;
      2'b01:
        casez_tmp = io_i_in_1;
      2'b10:
        casez_tmp = io_i_in_2;
      default:
        casez_tmp = io_i_in_3;
    endcase
  end // always_comb
  assign io_o_out = casez_tmp;
endmodule
```
Maybe there are other options to massage it into the form you want.

Alternatively, you can always implement dynamic indexing yourself. I have not texted this extensively but the following seems to do what you want:
```scala
object BinaryMuxTree {
  def apply[T <: Data](in: Seq[T], sel: UInt): T = {
    in.length match {
      case 1 => in.head
      case 2 => Mux(sel(0), in(1), in(0))
      case _ =>
        // Split vector in half
        val mid = 1 << (log2Ceil(in.length) - 1)
        val left = in.take(mid)
        val right = in.drop(mid)
                                                                                                                                                                                                                  
        val leftMux = apply(left, sel)
        val rightMux = apply(right, sel)

        Mux(sel(log2Ceil(in.length)-1), rightMux, leftMux)
    }
  }
}

class MuxTree(nBit: Int) extends Module {

  val io = IO(new Bundle {
    val i_slct = Input(UInt(3.W))
    val i_in = Input(Vec(4, UInt(nBit.W)))
    val o_out = Output(UInt(nBit.W))
  })

  io.o_out := BinaryMuxTree(io.i_in, io.i_slct)
}
```

The expressions are inlined but it has the tree structure you want:
```verilog
module MuxTree(
  input        clock,
               reset,
  input  [2:0] io_i_slct,
  input  [7:0] io_i_in_0,
               io_i_in_1,
               io_i_in_2,
               io_i_in_3,
  output [7:0] io_o_out
);

  assign io_o_out =
    io_i_slct[1]
      ? (io_i_slct[0] ? io_i_in_3 : io_i_in_2)
      : io_i_slct[0] ? io_i_in_1 : io_i_in_0;
endmodule
```

escou64, 2025-03-19:
Thank you, I will try these suggestions.
I think the Firtool option is more suitable: I would like to do that for any multiplexer of my design.
I can't find any documentation with all the existing options for Firtool. Is there an accessible link ?

jackkoenig, 2025-03-19:
> I think the Firtool option is more suitable: I would like to do that for any multiplexer of my design.

Can you explain why this particular emission matters to you? Is there a particular tool you are using that does not work well with the default emission?

> I can't find any documentation with all the existing options for Firtool. Is there an accessible link ?

I don't think there's a web page, but passing `--help` to firtool will tell you the options. You can do this when invoking it via Chisel:
```scala
ChiselStage.emitSystemVerilog(
  gen = ...,
  firtoolOpts = Array("--help")
)
```

escou64, 2025-03-21:
For the security analysis of a design, I need to check for each clock cycle the values of any RTL signal where a data can be propagated. In the case of a multiplexer tree, it leads to verifying all the potential connections between the multiplexers themselves.
Then, I need to break down all these constructs.

I will try to explore the existing options using the `--help` option, thank you.