chipsalliance/chisel #4847: Unable to import circt.stage.ChiselStage

tobias1012, 2025-04-01:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**
in any project import 
```
import circt.stage.ChiselStage
```
**What is the current behavior?**

error in sbt:
```
[error] C:\...\src\main\scala\proj\main.scala:5:22: object stage is not a member of package chisel3.util.circt
[error] import circt.stage.ChiselStage
```

**What is the expected behavior?**

Imports correctly so it is possible to generate verilog.

**Please tell us about your environment:**

- Version: 6.7.0
- OS: Windows and WSL, problem persists.
- sbt buildt tool

**Other Information**

**What is the use case for changing the behavior?**

seldridge, 2025-04-01:
Your code has a Scala error: there are two circt packages: `circt` and `chisel3.util.circt`. The likely problem is that your code does a wildcard import of `chisel3.util.circt` and then tries to import `circt.stage.ChiselStage`. Scala thinks you are trying to access `chisel3.util.circt.stage.ChiselStage` which does not exist. Two options:

1. Don't use a wildcard import of `chisel3.util._`. Then your code will work:

``` scala
import chisel3._
import chisel3.util.Counter
import circt.stage.ChiselStage
```

2. Keep the wildcard import and explicitly import explicitly from the root:

``` scala
import chisel3._
import chisel3.util._
import _root_.circt.stage.ChiselStage
```

tobias1012, 2025-04-04:
Hi @seldridge 
Thanks for the clarification, my code was the issue.

closed 2025-04-04 by tobias1012