chipsalliance/chisel #4877: Inaccrate Infomation in the "Get me Verilog" section of FAQ

scratch-er, 2025-04-09:
In the "Get me Verilog" section of the FAQ page (the links below), some information might be ambiguous, inaccurate, or outdated.

https://github.com/chipsalliance/chisel/blob/main/docs/src/resources/faqs.md#get-me-verilog

https://www.chisel-lang.org/docs/resources/faqs#get-me-verilog

The FAQ says that the following code generates a `HelloWorld.v` under PWD. However, `emitSystemVerilog` returns a string, and it is `emitSystemVerilogFile` that generates a file.

```scala
import circt.stage.ChiselStage
object VerilogMain extends App {
  ChiselStage.emitSystemVerilog(new HelloWorld)
}
```

The FAQ also suggests running the following command for fine control over verilog generation.

```sh
sbt 'runMain intro.HelloWorld --target-dir buildstuff --top-name HelloWorld'
```

However, the code spinet in the page does not contain an `intro.HelloWorld` object extending `App`, so this command will fail. Some code like this is needed for the above command to work.

```scala
object HelloWorld extends App {
    ChiselStage.emitSystemVerilogFile(new HelloWorld, args)
}
```

scratch-er, 2025-04-09:
Similar problems in the "Get me FIRRTL" section too.

seldridge, 2025-04-09:
Good catch, yes this should be updated.

This probably needs two things:

1. An example using scala-cli.
2. An example with a main function.

Anything related to `sbt` should be purged from (2).

If you have bandwidth for a PR, that would be great. Otherwise, I'll backlog this until I have time.

scratch-er, 2025-04-11:
> Good catch, yes this should be updated.
>
> This probably needs two things:
>
>     1. An example using scala-cli.
>
>     2. An example with a main function.
>
>
> Anything related to `sbt` should be purged [...]

I am a new Chisel user. I get confused by the inaccurate information in the docs when I am porting my Verilog design to Chisel. Eventually I realized the FAQ is not consistent with the behavior of the latest code, and I am posting an issue here.

I do have free time to help improve the docs. However, since I am a new Chisel user, if I am writing the docs, it might contain inaccurate information and code that is not best practice.

BTW, If I find some other problems in the docs, should I post a new issue or add these problems to this issue?

seldridge, 2025-04-11:
> I do have free time to help improve the docs. However, since I am a new Chisel user, if I am writing the docs, it might contain inaccurate information and code that is not best practice.

We'll review it, so that's no problem. Sometimes it is also better if the docs are written by someone not tightly connected to the project. Whatever you want to do here is fine.

> BTW, If I find some other problems in the docs, should I post a new issue or add these problems to this issue?

If you find new stuff, please open new issues. 👍

closed 2026-11-16 by jackkoenig