chipsalliance/chisel #4899: How to parse firrtl file/string to Circuit instance?

canxin121, 2025-04-20:
I can convert a Circuit instance into string FIRRTL code using Serializer.serialize(circuit). So, how can I do the reverse: parse FIRRTL code (or a file) back into a Circuit instance?

I asked AI, and it suggested `val circuit = Parser.parseFile(inputFile)`. However, the Parser object actually no longer has this parseFile method. Instead, the file as follows:

``` scala
package firrtl

object Parser {

  sealed abstract class InfoMode

  case object IgnoreInfo extends InfoMode

  case object UseInfo extends InfoMode

  case class GenInfo(filename: String) extends InfoMode

  case class AppendInfo(filename: String) extends InfoMode

}
```

seldridge, 2025-04-20:
Can you clarify on what you are trying to do?

Support for this was dropped when migrating to CIRCT several years ago. That last version that would have allowed you to do this was Chisel 3.6 and specifically, the corresponding FIRRTL 1.6 Scala project had an (auto-generated) FIRRTL parser in Scala.

If you're trying to parse FIRRTL, the right project for that is the CIRCT project (https://github.com/llvm/circt) and specifically it's `firtool` or `circt-translate` tools. Using that, you can parse FIRRTL text into FIRRTL Dialect with `firtool -parse-only Foo.fir` or `circt-opt -import-firrtl Foo.fir`.

canxin121, 2025-04-21:
> Can you clarify on what you are trying to do?
>
> Support for this was dropped when migrating to CIRCT several years ago. That last version that would have allowed you to do this was Chisel 3.6 and [...]

I have written a Chisel Phase and want to apply them to FIRRTL files or strings. 

```scala
class CoverageTransform extends firrtl.options.Phase { ... }
```

Now I can create my own Stage, similar to ChiselStage, and include my own Phases within it to transform the Module and generate SystemVerilog.

```scala
    val stage = new CustomStage(
      customPhases = Seq(new CoverageTransform)
    )
    val transformed_sv = stage.emitSystemVerilog(
      new UART_tx()
    )
```

But the problem is, some Chisel modules are supplied only as FIRRTL files. It's difficult to include the original Chisel source code because of complex dependencies. In this scenario, I don't know how I should perform the Phase transformations on these FIRRTL files.

For example, I have a uart_tx.fir file as below that needs to be transformed using the CoverageTrans Phase. How should I do this?

```firrtl
FIRRTL version 3.3.0
circuit UART_tx :
  module UART_tx : 
    input clock : Clock 
    input reset : UInt<1> 
    output io : { flip i_tx_trig : UInt<1>, flip i_data : UInt<8>, o_tx_busy : UInt<1>, o_tx_done : UInt<1>, o_serial_data : UInt<1>} 
    ...
```