chipsalliance/chisel #4979: svsim.Simulation$UnexpectedEndOfMessages$ when running SimulatorSpec with custom VcsSimulator

tututtt, 2025-07-16:
**Type of issue**: Bug Report

**Please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**Other Information**

**What is the use case for changing the behavior?**

seldridge, 2025-07-16:
That's odd, in RC1 I see a compilation failure:

```
[367] Error-[SFCOR] Source file cannot be opened
[367]   Source file "layers-GCD-Verification.sv" cannot be opened for reading due to
[367]   'No such file or directory'.
[367]   Please fix above issue and compile again.
[367]   "../primary-sources/verification/assert/layers-GCD-Verification-Assert.sv", 
[367]   4
[367]   Source info:   `include "layers-GCD-Verification.sv"    // 
[367]   src/test/scala-2/chiselTests/simulator/GCD.scala:6:7
[367] 
[367] 1 error
```

However, if I run on RC2, things compile fine, but there is a problem with the test (which is handled gracefully) documented in https://github.com/chipsalliance/chisel/issues/4980. Can you try on RC2 and see if the message is gone?

`UnexpectedEndOfMessage` really shouldn't show up anymore as the way that ChiselSim works is that it will always post-process log files looking for what happened on any failure. Previously, you would have to catch this exception yourself. However, this was fixed before RC1.

tututtt, 2025-07-18:
> It looks like a VCS version issue. The problem is that VCS will do some weird stuff with address space layout randomization (ASLR). By default, if VCS detects that ASLR is running, it shuts down and restarts itself. This is necessary to enable it's "save/restore" functionality. The easiest way to disable this (and not confuse ChiselSim) is to use the `-no_save` option. There used to be a much more complicated workaround for this that I removed and changed to `-no_save`: [5415546](https://github.com/chipsalliance/chisel/commit/5415546a6885f532bbf8c6e89a2184952cf15ea5) (Noting that this complicated workaround didn't work for some users either if, I think, they didn't have ASLR enabled.) However, the problem is that with earlier versions of VCS, they do not support `-no_save`.
> 
> The smoking gun was looking in `workdir-vcs/simulation-log.txt` and seeing that there was junk in there:
> 
> ```
> r ready                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       
> e Could not read next command.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
> nsupported option '-no_save' is ignored                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       
>                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               
> Chronologic VCS simulator copyright 1991-2017                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 
> Contains Synopsys proprietary information.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    
> Compiler version M-2017.03-SP2_Full64; Runtime version M-2017.03-SP2_Full64;  Jul 17 08:16 2025                                                                                                                                                                                                                                                                                                                                                                                                                                                             
> ```
> 
> VCS is not handling the `-no_save`, it dumps junk into the file, then this is a true "UnexpectedEndOfMessage" as I think VCS has already shut down (or it is failing to parse the junk above).
> 
> I _can_ reproduce your error on:
> 
> * `M-2017.03-SP2`
> 
> I _cannot_ reproduce this on (the test works on):
> 
> * `V-2023.12`
> 
> Can you use a newer version of VCS?
> 
> You may try disabling `-no_save` here:
> 
> [chisel/svsim/src/main/scala/vcs/Backend.scala](https://github.com/chipsalliance/chisel/blob/5b886b6a553d6cd23b83922afed103980b3c0f74/svsim/src/main/scala/vcs/Backend.scala#L423)
> 
> Line 423 in [5b886b6](/chipsalliance/chisel/commit/5b886b6a553d6cd23b83922afed103980b3c0f74)
> 
>  // In order to support save/restore functionality, VCS will detect if 
> Note: I am _not_ able to get this to work by disabling `-no_save` on my end. If I do that, I have the problem where VCS is doing it's "double start" ASLR disable hack and that _also_ corrupts the messages/produces an `UnexepctedEndOfMessage` error.

Thank you for the explanation. I attempted to disable the -no_save option, but unfortunately the UnexpectedEndOfMessage error still persists. I’ll proceed with updating to a newer version of VCS, as that appears to be the more stable path forward.

closed 2025-07-19 by tututtt