chipsalliance/chisel #4984: Chisel asserts should give file and line number to Chisel as well as systemverilog

oharboe, 2025-07-19:
**Type of issue**: Feature Request

```
  chisel3.simulator.Exceptions$AssertionFailed: One or more assertions failed during Chiselsim simulation
------------------------------------------------------------------------------
The following assertion failures were extracted from the log file:

  lineNo  line
  ----------------------------------------------------------------------------
       0  [21] %Error: XxxTestBench_Verification_Assert.sv:301: Assertion failed in TOP.svsimTestbench.dut.verification_assert.unnamedblk1: Assertion failed: Must be positive

For more information, see the complete log file:

  test_run_dir/aptos.Simulator_889627/workdir-default/simulation-log.txt
```

**Is your feature request related to a problem? Please describe.**

I have to dig into the XxxTestBench_Verification_Assert.sv file to find the Chisel .scala file and line number

**Describe the solution you'd like**

Print out the .scala filename and line number of the assert

seldridge, 2025-07-19:
This seems reasonable. I think in the past we've handled this by having the assertion auto-insert into its message all the source locator information.

What I would rather do here is to start using explicit `else` blocks for assertions to get an emission like:

``` systemverilog
assert property(...)
else
  $display("Scala source line: ???")
```

At least for our flows we would need a way of stripping this additional information so we don't ship it to a customer. There may be a construction that uses layers to add this information and allow it to be easily stripped.

oharboe, 2025-07-19:
I was thinking this would be a feature of assert.

Are you suggesting the user should modify all asserts to get this information or that this is an implementation detail of assert?

seldridge, 2025-07-19:
Yeah, it would be a feature of assert to do this automatically for you. I wouldn't suggest users do this manually.

I think `chisel3.assert` will already do this, but we're generally trying to move away from that towards `ltl.AsertProperty` which does _not_ do this as it doesn't have any print associated with it.